bdring/Grbl_Esp32 #760: New PCB boards

rlwoodjr, 2021-02-09:
I see there are several circuit board that you can purchase for theESP32 grbl firmware.  

I would like to have a board capable of 5VDC inputs and outputs and have screw terminals for the inputs and outputs.

bdring, do you make custom boards? If so, what type of volume is required to create a custom board? 

Or, where is a good place to have one made?

bdring, 2021-02-10:
I sell some via Tindie as a part time project. I retired some of the original controllers and primarily sell the [6 Pack](https://www.tindie.com/products/edit/6-pack-universal-cnc-controller/) now. It can handle 5V I/O.

I have a few other designs coming on line soon. They are being assembled right now. 

[Join our Discord server](https://discord.gg/de8HzaXMGG) to see the status of the new controllers and discuss new ideas.

rlwoodjr, 2021-02-10:
Please resend the invite to join the Discord server.  I got distracted and it expired.

bdring, 2021-02-10:
Hmmm. It was supposed to be a never expire 50 use invite.

Sorry, Trying again.

https://discord.gg/ZbgjKvNVut

rlwoodjr, 2021-02-10:
I get this message

[image]

rlwoodjr, 2021-02-10:
If I continue, it gives me a dialog box but you cannot type in it.

rlwoodjr, 2021-02-10:
I tried again and it worked. Not sure what happened.  Thanks.

closed 2021-02-10 by rlwoodjr