bdring/Grbl_Esp32 #761: PWM Laser/servo mode change

maximius, 2021-02-10:
Good afternoon, Bart.
Thank you very much for your work. I want to build a Drawing Robot (CoreXY mechanics), based on your controller. But there was one question about working with servo.

I use the GRBL v1.3a firmware. Configuration of pen_laser. h. I configured the laser control successfully(on the PWM pin with a frequency of 5 kHz, the duty cycle changes when the parameter S changes within 0-255)

When I select #define USE_PEN_SERVO in config.h. in the configuration, the pulse frequency on the PWM pin becomes 50 Hz (pulse length 2ms), but when changing the position of the servo (for example, m3 s10 or m3 s20), the pulse length does not change. and the pulses start to be generated only when the axes are moved. I tried different configurations and settings, but I didn't learn how to change the position of the servo.
I ask for help in this matter. How to properly control the servo to move the handle? Is it possible to quickly switch from the laser mode to the servo mode? Otherwise, you have to have two ESP32 modules.
While using GRBL_Servo on the arduino Uno and to control the servo, the commands M3 S20 for moving down and m5 for lifting the handle are used.

bdring, 2021-02-10:
The servo is used as a motor for an axis. You control it by moving the axis. For a pen machine you would typically set it up as the Z axis. It has nothing to do with a spindle.

Be sure you are using the latest build firmware and use [this Wiki page](https://github.com/bdring/Grbl_Esp32/wiki/Servo-Axis-Feature) for reference.

maximius, 2021-02-10:
I am just studying this page and have already learned how to move the servo.
If I like something to change the limits. By default, the console writes the following settings:
[[MSG:Z Axis RC Servo Pan:27 PulseLen(3276,6553) Limits(-100.000,0.000)]
Can I change the limits? By changing the $$ parameters for the Z axis, the settings have not changed
I also noticed the peculiarity that the position of the servo depends on the machine coordinates Zm, and not on the working Zw. Although running programs change the Zw. Thus, for the correct operation of the servo, we need to first reset Zm in the control program, so that Zw is equal to Zm, and only then work with this axis. Do I understand the logic of the work correctly?
And is it possible to switch pin27 from working with a servo to working with a PWM laser without updating the firmware? Viewing the list of $L did not give me an answer. I want to use both a laser and a pen on the plotter, but I don't want to load ESP 32 through the Arduino IDE every time. So far, I'm going to use 2 ESP 32 modules with different programs loaded, but this is wrong, as I believe.

bdring, 2021-02-10:
The servo range works in machine space. [Here is a wiki page](https://github.com/bdring/Grbl_Esp32/wiki/Machine-Space-and-Homing) on how to set that up. You can create work space offset within that area, but keep in mind the servo will be limited to the machine space you setup.

You can use different pins for the servo and laser. You could even use them at the same time. You must select those at compile time, but you should be able to create a single setup for both modes. 

Please spend some time reading the wiki, then [join us on Discord](https://discord.gg/ybzepxRUBK) if you have more basic setup questions.

maximius, 2021-02-11:
Thank you for the information. I will study the wiki again - I see that there are a lot of new things there.

closed 2021-02-11 by maximius