bdring/Grbl_Esp32 #762: Not resuming after Door closed

AlphaCircuit, 2021-02-10:
What version of the firmware are you using? 1.3a 20201212

Is the problem repeatable?Yes

Under what conditions does the bug occur?Opening door and closing door

Problem every time I open the door the system reports info for "?' request as below but ignores all "$'commands and refuse to resume after door closed. ie it should resume but goes into a locked state can only poll "?" everything else is no response.

<Door:0|MPos:0.000,0.000,0.000|FS:0,0|WCO:0.000,0.000,0.000>     **Note door state but it still not responding to $ command non responsive to everything like in a coma only can poll "?"**
[MSG:Check door]
<Door:1|MPos:0.000,0.000,0.000|FS:0,0|Pn:D|Ov:100,100,100>     **Open door to trigger problem**

No code because I am not sure where the problem stems from.

bdring, 2021-02-10:
Please upgrade to the current code.

The door will trigger the door state and status will report **Door:1**. That is a door state with door open still reporting. You need to close the door. It will then report the state as **Door:0**. You then need to send the cycle start command with is the **~** character. That will put you back into the mode to control the machine.

We are actively talking about what you should be allowed to do in the various modes on the Discord server. It is rather conservative now. It takes on more significance with the next major release of Grbl_ESP32. That is a compile once type firmware and you can define things on the fly. If your door circuit is not working and stuck in open mode, you should be able to undefine the door to fix it.

This will take some time to do a correct and comprehensive enhancement to the current condition.

AlphaCircuit, 2021-02-11:
Yeah my bad... after reading your reply realized I need to use the cycle start ... problem was me .... Thanks bdring...

closed 2021-02-20 by bdring