bdring/Grbl_Esp32 #763: Less I/O ports than buttons

JensHauser, 2021-02-11:
Is there some possibility to add 2 macros/features to 1 button?

Approaches I have in mind:
- Single press, double press the button
- Short press, long press the button

"Hold" and "Resume" would be the first I´d like to put on one button.

bdring, 2021-02-11:
No. but with custom code you can use logic to control the "macro".

It would check the state and apply the action. if (state = hold) then resume.

bdring, 2021-02-11:
See

https://github.com/bdring/Grbl_Esp32/wiki/Macro-Button-Feature#advanced-macros

JensHauser, 2021-02-11:
Ok, have the skeleton of the user_defined_macro.
The grbl_sendf function is called upon button press and prints a message in the web interface

4axis_xyxz.h
```
#pragma once
#define MACHINE_NAME            "MACHINE_ESP32 Jens XYZA"
#define CUSTOM_CODE_FILENAME    "Custom/4axis_xyxz.cpp"

#ifdef N_AXIS
    #undef N_AXIS
#endif

#define N_AXIS 3
#define SPINDLE_TYPE SpindleType::RELAY
#define SPINDLE_OUTPUT_PIN      GPIO_NUM_32

// Das 2. Relais
#define COOLANT_FLOOD_PIN       GPIO_NUM_2
#define STEPPERS_DISABLE_PIN    GPIO_NUM_13 //ok

#define X_STEP_PIN              GPIO_NUM_12
#define X_DIRECTION_PIN         GPIO_NUM_14
#define X_LIMIT_PIN             GPIO_NUM_17

#define Y_STEP_PIN              GPIO_NUM_26
#define Y_DIRECTION_PIN         GPIO_NUM_15
#define Y_LIMIT_PIN             GPIO_NUM_4

#define Z_STEP_PIN              GPIO_NUM_27
#define Z_DIRECTION_PIN         GPIO_NUM_33
#define Z_LIMIT_PIN             GPIO_NUM_16

#define LIMIT_MASK          B111

#define PROBE_PIN               GPIO_NUM_35 //ok
#define CONTROL_RESET_PIN       GPIO_NUM_34  // needs external pullup
#define CONTROL_CYCLE_START_PIN GPIO_NUM_39  // needs external pullup
#define MACRO_BUTTON_0_PIN      GPIO_NUM_36

// wrap prototypes
#ifndef FourAxis_xyxz_h
    #define FourAxis_xyxz_h

    void user_defined_macro(uint8_t index);
#endif
```

4axis_xyxz.cpp
```
#include <unistd.h>
#include "../Machines/4axis_xyxz.h"

#define DEBUG // do I want debug messages? yes/no

#if defined(MACRO_BUTTON_0_PIN)
void user_defined_macro(uint8_t index) {
	#ifdef DEBUG
		grbl_sendf (CLIENT_ALL, "Macro button #%d pressed. State=%d \r\n", index, sys.state);
	#endif

	if (index == 0) { // Macro Button 1 pressed
		switch (sys.state) {
			case State::Cycle:
				// Set "cycle mode"
				break;

			case State::Hold:
                                // Set "hold mode"
				break;

			default:
				break;
		}
	}
}
#endif
```

Issues, kindly ask for help
(1)
Quick button press means sometimes doubled messages. I guess that´s a debouncing issue.
Debouncing via #define ENABLE_SOFTWARE_DEBOUNCE is, according to the sources, not applicable for this button. It would be helpful to just wait a few milliseconds right before returning from that function. If that was a feasible solution, how to do that?

```
<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:-205.000,205.000,-55.000>
Macro button #0 pressed. State=0 
Macro button #0 pressed. State=0 
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
```

(2)
In the wiki page I read that a char[] with regular G code seems to be the wrong approach, because the state needed to be "idle".
How to set the "hold" & "resume" state? In the Atari or other files I couldn`t find examples.

Thanks so much for your help

bdring, 2021-02-11:
You should join us on Discord for stuff like this.

https://discord.gg/ZbgjKvNVut

JensHauser, 2021-02-12:
Works now, thanks to Bart:

The debouncing issue better requires a hardware solution, the adjusted sources are here

```
#include <unistd.h>
#include "../Machines/4axis_xyxz.h"

#define DEBUG // do I want debug messages? yes/no

#if defined(MACRO_BUTTON_0_PIN)
void user_defined_macro(uint8_t index) {
	#ifdef DEBUG
		grbl_sendf (CLIENT_ALL, "Macro button #%d pressed. State=%d \r\n", index, sys.state);
	#endif

	if (index == 0) { // Macro Button 1 pressed
		switch (sys.state) {
			case State::Cycle:
				// Set "cycle mode"
				sys_rt_exec_state.bit.feedHold = true;
				break;

			case State::Hold:
                                // Set "hold mode"
				sys_rt_exec_state.bit.cycleStart = true;
				break;

			default:
				break;
		}
	}
}
#endif
```

closed 2021-02-12 by JensHauser

JensHauser, 2021-02-13:
Works even better, now with 2 macro buttons

- Button #1 is hold and resume (I saved one button). Applicable only during cycling
- Button #2 does some homing and then moves to a convenient tool change position

The whole function is backed up with non-blocking software debouncing.

```
#define DEBOUNCE_TIME_MACRO_1 300  // ms, my button is heavily bouncing. verified with oscilloscope

#include <stdint.h>
#include "src/Grbl.h"
#include "Machines/4axis_xyza.h"

uint32_t earlier = 0;

#if defined(MACRO_BUTTON_0_PIN) || defined(MACRO_BUTTON_1_PIN)
void user_defined_macro(uint8_t index) {
    uint32_t later, msPassedBy = 0;

    later      = millis();
    msPassedBy = later - earlier;

    if (msPassedBy >= DEBOUNCE_TIME_MACRO_1) {
        switch (index) {
            case 0:  // Macro button 1 (Hold/Cycle switch)

                switch (sys.state) {
                    case State::Hold:
                        grbl_sendf(CLIENT_ALL, "Macro button #%d pressed. New state \"Cyle start\"\r\n", index + 1);
                        sys_rt_exec_state.bit.cycleStart = true;
                        break;

                    case State::Cycle:
                        grbl_sendf(CLIENT_ALL, "Macro button #%d pressed. New state \"Hold\"\r\n", index + 1);
                        sys_rt_exec_state.bit.feedHold = true;
                        break;

                    default:
                        grbl_sendf(CLIENT_ALL, "Macro button #%d pressed. Works only in states \"Cycle\" or \"Hold\"\r\n", index + 1);
                        break;
                }
                break;

            case 1:  // Macro button 2 (Homing)
                if (sys.state == State::Idle) {
                    grbl_sendf(CLIENT_ALL, "Macro button #%d pressed. Homing, Y tool change position\r\n", index + 1);

                    grbl_sendf(CLIENT_ALL, "$H\r\n");
                    WebUI::inputBuffer.push("$H\r\n");  // Homing all axis

                    grbl_sendf(CLIENT_ALL, "G53 G0 Z-5\r\n");
                    WebUI::inputBuffer.push("G53 G0 Z-5\r\n");  // Move Z axis up (should already be there, just to be sure)

                    grbl_sendf(CLIENT_ALL, "G53 G0 X-5 Y-210 F1000\r\n");
                    WebUI::inputBuffer.push("G53 G0 X-5 Y-210 F1000\r\n");  // Move Y axis to the middle for tool change
                } else {
                    grbl_sendf(CLIENT_ALL, "Macro button #%d pressed. Works only in state \"Idle\"\r\n", index + 1);
                }
                break;

            default:
                break;
        }

        earlier = later;
    }
}
#endif
```