bdring/Grbl_Esp32 #766: $4 Step enable invert - not working

mtk11, 2021-02-12:
Config:

[MSG:Grbl_ESP32 Ver 1.3a Date 20210204]
[MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
[MSG:Using machine:ESP32 I2S 5 Axis Driver Board]
[MSG:Axis count 5]
[MSG:I2S Steps, Stream]
[MSG:Init Motors]
[MSG:X  Axis Standard Stepper Step:I2SO(1) Dir:I2SO(0) Disable:I2SO(2) Limits(0.000,300.000)]
[MSG:Y  Axis Standard Stepper Step:I2SO(4) Dir:I2SO(3) Disable:I2SO(5) Limits(0.000,300.000)]
[MSG:Z  Axis Standard Stepper Step:I2SO(7) Dir:I2SO(6) Disable:I2SO(8) Limits(-300.000,0.000)]
[MSG:A  Axis Standard Stepper Step:I2SO(10) Dir:I2SO(9) Disable:I2SO(11) Limits(-300.000,0.000)]
[MSG:B  Axis Standard Stepper Step:I2SO(13) Dir:I2SO(12) Disable:I2SO(14) Limits(-300.000,0.000)]
[MSG:CoreXY Kinematics Init]

Tried changing $4 via the serial and web interface, it shows the value change, but the step-motor driver pin continue to work the same way.

bdring, 2021-02-12:
Yes, I think we know about this one. It only affects systems with separate enable pins for each motor. I'll give it a look. It should be an easy fix.

bdring, 2021-02-12:
Try the [EnablePerMotorFix](https://github.com/bdring/Grbl_Esp32/tree/EnablePerMotorFix) branch.

It is based on the Devt branch. If it works for you, I will create a P.R.

mtk11, 2021-02-12:
EnablePerMotorFix branch works, thanks. 

Some other issues appeared: it seems to double the step-motor steps, Let say if move 10 mm, the actual movement is 20 mm. Also, when the carriage suppose to move with constant speed some rapid speed increase and decrease were observed, kind of randomly. I am using LV8729 @ 128 µStep and had similar issues with Marlin in the past. It worked with I2S clock frequency increase and decrease of data transmission length to 16-bit. To try the same ?

bdring, 2021-02-13:
Not sure why that happened. The code change was very subtle and should not have affected anything else. I just added  a small change to reduce excess setting of the same value to the pin. That should not affect your situation because that part of the code was always there.

Are you sure you are using the correct machine definition? Can you try again with the latest change and paste the startup messages in a reply.

mtk11, 2021-02-13:
I copied from [EnablePerMotorFix](https://github.com/bdring/Grbl_Esp32/tree/EnablePerMotorFix) branch [Motors.cpp](https://raw.githubusercontent.com/bdring/Grbl_Esp32/9122d4c2b16f41a5bbd8efce4acc7ec4011c37bc/Grbl_Esp32/src/Motors/Motors.cpp) to [CoreXYFixes](https://github.com/bdring/Grbl_Esp32/tree/CoreXYFixes) branch and also removed #define MIDTBOT  - it seems to work.

Other issue – G3 code. When execute, let say:

G3X10.51Y18.47I-3.94J0F200

The feedrate seems not properly executed. At the beginning it adjusts the speed, but short after the speed increases.

bdring, 2021-02-13:
Is your machine a CoreXY?
Can you give the position right before the G3 command?
Can you paste the [boot messages](https://github.com/bdring/Grbl_Esp32/wiki/Requesting-Help#boot-messages) here?

mtk11, 2021-02-13:
> Is your machine a CoreXY?

CoreXY [Drawing Machine](https://www.thingiverse.com/thing:1514145), machine config:

#define CUSTOM_CODE_FILENAME    "../Custom/CoreXY.cpp"
#define USE_KINEMATICS      // there are kinematic equations for this machine
#define USE_FWD_KINEMATICS  // report in cartesian

> Can you give the position right before the G3 command?

The position right before the G3 command:

<Idle|MPos:0.000,0.000,0.000,0.000,0.000|FS:0,0>
G0X10.51Y18.47
ok
<Idle|MPos:10.510,18.470,0.000,0.000,0.000|FS:0,0|Ov:100,100,100>
....
G3X10.51Y18.47I-3.94J0F200
ok
<Idle|MPos:10.510,18.470,0.000,0.000,0.000|FS:0,0>

> Can you paste the [boot messages](https://github.com/bdring/Grbl_Esp32/wiki/Requesting-Help#boot-messages) here?

[MSG:Grbl_ESP32 Ver 1.3a Date 20210204]
[MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
[MSG:Using machine:ESP32 I2S 5 Axis Driver Board]
[MSG:Axis count 5]
[MSG:I2S Steps, Stream]
[MSG:Init Motors]
[MSG:X  Axis Standard Stepper Step:I2SO(1) Dir:I2SO(0) Disable:I2SO(2) Limits(0.000,200.000)]
[MSG:Y  Axis Standard Stepper Step:I2SO(4) Dir:I2SO(3) Disable:I2SO(5) Limits(0.000,200.000)]
[MSG:Z  Axis Standard Stepper Step:I2SO(7) Dir:I2SO(6) Disable:I2SO(8) Limits(-300.000,0.000)]
[MSG:A  Axis Standard Stepper Step:I2SO(10) Dir:I2SO(9) Disable:I2SO(11) Limits(-300.000,0.000)]
[MSG:B  Axis Standard Stepper Step:I2SO(13) Dir:I2SO(12) Disable:I2SO(14) Limits(-300.000,0.000)]
[MSG:CoreXY Kinematics Init]
[MSG:Laser spindle on Pin:GPIO(27), Enbl:None, Freq:5000Hz, Res:13bits Laser mode:On]

[MSG:Client Started]
[MSG:Connecting xxxxxxxxxx]
.......
[MSG:Start mDNS with hostname:http://grblesp.local/]
[MSG:SSDP Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]
[MSG:X  Axis limit switch on pin GPIO(35)]
[MSG:Y  Axis limit switch on pin GPIO(32)]
[MSG:Z  Axis limit switch on pin GPIO(33)]

bdring, 2021-02-14:
I can try it on a CoreXY machine. 
Is this on hardware you designed? (ESP32 I2S 5 Axis Driver Board)?

mtk11, 2021-02-14:
> I can try it on a CoreXY machine.

On lower feedrates is more noticeable.

> Is this on hardware you designed? (ESP32 I2S 5 Axis Driver Board)?

A slightly modified version of @simon-jouet [ESP32Controller R2.](https://github.com/simon-jouet/ESP32Controller/tree/r2) 

[image]

I am considering making something smaller for such compact machines. Not sure if ESP32-S2 or ESP32-S3 might be more appropriate ?

bdring, 2021-02-14:
We have a controller design channel on our [Discord server](https://discord.gg/Nzz9gWZV4K).

bdring, 2021-02-14:
It seems to run smoothly on my CoreXY machine. I have an AntPCB Maker, I use to test. I also tried slow feed rates and accelerations.

I don't know how the enable changes could have affected anything. It is more likely a CoreXY issue, but I am not seeing it.

CoreXY has to adjust the feed rate of the motors to keep the end effector moving at a constant rate as the angle of motion changes. MidTbot requires a different calculation because the motors are moving.

mtk11, 2021-02-14:
> It is more likely a CoreXY issue, but I am not seeing it.

Just tried DRV8825, the same issue. It is quite noticeable with the following code (generated with [LaserGRBL](https://lasergrbl.com/)):

G90 (use absolute coordinates)
M3 S0
S0 
G0X1.87Y1.64
S1000 
G1X11.83F200
Y11.6
X1.87
Y1.64
S0 
G0X10.51Y18.47
S1000 
G3X10.51Y18.47I-3.94J0F200
S0 
M5 S0
G0 X0 Y0 Z0 (move back to origin)

It draws the square properly at F200, while the circle - with feedrate like F1000.

bdring, 2021-02-14:
OK can see the problem. I ran the same code on a CoreXY machine and a standard machine. The problem only occurs on the CoreXY.

I will investigate. It does not appear to be random. It appears to be using the rapid rate. It is smooth for me, just faster than it should be. Try changing the$X/Maxrate=200 (and Y) to match your F200 feed rate. 

I don't think the CoreXY kinematics is using the right feed rate. It is likely due to the extra steps involved with an arc.

bdring, 2021-02-14:
I think i found the problem. The CoreXY kinematics is altering the feed rate like it should, but the loop in the mc_arc(...) function is reusing the altered value and it grows each time in the CoreXY function.

I added 2 lines. Could you try making these changes and test? 

[image]

bdring, 2021-02-14:
It has been added to the  EnablePerMotorFix branch

mtk11, 2021-02-14:
> I added 2 lines. Could you try making these changes and test?

So far it works fine, no other issues were found.

Should I submit the machine config I am using for [ESP32Controller R2](https://github.com/simon-jouet/ESP32Controller/tree/r2), in case someone else uses this design ?

bdring, 2021-02-14:
We have to keep the machine definition count down or we fail the automated test due to taking to long. We compile every one.

Do you have a fork? We could point people there.

mtk11, 2021-02-14:
> Do you have a fork? We could point people there.

Sorry, I am new to github, it created P.R. ##772, the file: [ESP32Controller_R2.h](https://github.com/bdring/Grbl_Esp32/pull/772/commits/525e594be4c5fe8301871aabcdf16ffc64a59f5b#diff-c2326d60dae5b958968faf387b4eb19586f3712e51f3e85cc797ea14f90381c3).

mtk11, 2021-03-28:
Another issue appeared. When lasering line by line, after few lines, there is a shift in the main direction. The issue appear both with horizontal and vertical raster.

[image]

bdring, 2021-03-29:
Please don't reopen an issue for a different topic.
If you are using I2S steps, try changing it to static mode.

```
// I2S (steppers & other output-only pins)
#define USE_I2S_OUT
#define USE_I2S_STEPS
#define DEFAULT_STEPPER ST_I2S_STATIC
```

closed 2021-03-29 by bdring