bdring/Grbl_Esp32 #767: Over the air updated

JensHauser, 2021-02-13:
I´m curious how to get OTA working. I tried it with platformIO 

[env]
upload_protocol = espota
upload_port = 192.168.xx.xx

But no luck.

```
CURRENT: upload_protocol = espota
Uploading .pio\build\release\firmware.bin
13:22:31 [DEBUG]: Options: {'esp_ip': '192.168.xx.xx', 'host_ip': '0.0.0.0', 'esp_port': 3232, 'host_port': 24402, 'auth': '', 'image': '.pio\\build\\release\\firmware.bin', 'spiffs': False, 'debug': True, 'progress': True, 'timeout': 10}
13:22:31 [INFO]: Starting on 0.0.0.0:24402
13:22:31 [INFO]: Upload size: 1230368
Sending invitation to 192.168.xx.xx
13:22:31 [INFO]: Waiting for device...
13:22:41 [ERROR]: No response from device
*** [upload] Error 1
```

Any ideas?

JensHauser, 2021-02-13:
It´been the Windows PCs local firewall, works now :-)

```
Uploading: [============================================================] 99%
Uploading: [============================================================] 99%
Uploading: [============================================================] 100% Done...

13:31:35 [INFO]: Waiting for result...
13:31:36 [INFO]: Result: OK
13:31:36 [INFO]: Success
```

bdring, 2021-02-13:
Nice, we should put this on the Wiki somewhere. 

I am not sure where though. We would probably need a full PlatformIO page. I have been afraid to do that because it would end up being one more thing to provide support on.

JensHauser, 2021-02-13:
If you want, I could draft a page, e.g. at first with headlines for alignment.

bdring, 2021-02-13:
Sure, that would be helpful.

JensHauser, 2021-02-13:
Some ideas about the PlatformIO wiki page (I´d suggest to add some pictures. That could be helpful for beginners)

1. Advantages of PlatformIO in comparison with Arduino IDE
2. How to compile Grbl_Esp32
2. Installation of Visual Studio Code
2. How to start with PlatformIO
2. Installation of PlatformIO and extensions
3. Setting up the project
4. Clean, Build, Upload (USB and over the air)
5. Update Grbl_Esp32

Unsorted ideas
- Update ESP32 firmware
- Fork Grbl_Esp32 and maintain own programming with GitHub
- FAQ (regular issues and problem solving)
- Simple program, e.g. small user defined function (header, .c file)

JimInAshland, 2021-02-13:
>    3. Installation of Visual Studio Code
>    4. How to start with PlatformIO
>    5. Installation of PlatformIO and extensions
>    6. Setting up the project
>    7. Clean, Build, Upload (USB and over the air)
>    8. Update Grbl_Esp32

As a rank beginner that would be very valuable.
Thank You for all You do!
Jim Chamberlain

MitchBradley, 2021-02-13:
@JensHauser Great list!  Please create such a page.

JensHauser, 2021-02-14:
Ok, what´s the regular cooking recipe to work out such a wiki site? Where to store the wiki page and pics, how to review with peers?

bdring, 2021-02-14:
One way to handle the pictures is to close this issue and just start pasting pictures here. That gives you a stable URL to reference.

I am not sure how to handle the review. If you fork this repo and add the changes at your fork, we could look at it there.

Github wikis are not very collaborative yet.

MitchBradley, 2021-02-14:
One way to review a wiki page is to publish it and let people make changes.

JensHauser, 2021-02-14:
[PlatformIO](https://github.com/bdring/Grbl_Esp32/wiki/Compiling-with-PlatformIO) wiki page set up

closed 2021-02-14 by JensHauser

JimInAshland, 2021-02-14:
THANK YOU SO MUCH