bdring/Grbl_Esp32 #768: WebUI, spindle button

JensHauser, 2021-02-13:
My machine uses a relay type spindle.

```
#define SPINDLE_TYPE SpindleType::RELAY
#define SPINDLE_OUTPUT_PIN      GPIO_NUM_32
```

In the WebUI, there is a button to manually switch it on/off, but that doesn´t work.
It´s just an issue in relation to the WebUI, in g code everything is fine.

Another relay is defined by and mapped, works perfectly by pressing the "FLOOD" button.
`#define COOLANT_FLOOD_PIN       GPIO_NUM_2`

Any hints?

luc-github, 2021-02-13:
spindle button send 0x9E command :  https://github.com/bdring/Grbl_Esp32/blob/c950e2a16d4c7c6ad9a3fe050cb2c9a6454abfa3/Grbl_Esp32/src/Config.h#L197

bdring, 2021-02-13:
All those commands are overrides.

The spindle button with a target in middle is to turn of the spindle in hold. I think there is an issue with using since we added spindle delays. I think our recent fix in a branch fixes it....checking

JensHauser, 2021-02-13:
The above was tested in IDLE mode

bdring, 2021-02-13:
Yes, the Parking_Delay_fix branch fixes the issue. I will create a P.R. for that into the Devt branch. I was waiting to make the parking parameters $ options, but that turned into a bigger project.

The WebUI is a separate project from Grbl_ESP32. 

What you are trying to do is typically done with the macro buttons.