bdring/Grbl_Esp32 #774: template.h file appears to be missing

UniqueLemming, 2021-02-17:
I can't seem to find the template.h file that the Wiki says should be in the Machines directory. Has the info from this file been moved elsewhere?

JensHauser, 2021-02-17:
Which wiki page are your referring to (link)?

I adjusted the "Custom Machine Functions" wiki with current link to the Machine folder

UniqueLemming, 2021-02-17:
It's in the "Setting up the I O Pins" section under the heading "Making Your Machine Definition File"

https://github.com/bdring/Grbl_Esp32/wiki/Setting-Up-the-I-O-Pins

JensHauser, 2021-02-17:
It´s been gone with pull [#732](https://github.com/bdring/Grbl_Esp32/pull/732). While this in clarification, you can take the last [version](https://github.com/bdring/Grbl_Esp32/blob/f7650799393bf8e5612868d6bfe5964af8f77219/Grbl_Esp32/src/Machines/template.h)

bdring, 2021-02-20:
template.h was removed a while ago. It was a basic template for creating a machine definition file. The .h file was not a good format for doing that. Markdown (wiki) would be better. It was not maintained well and causing more questions than it was answering.

There are no plans to replace it in the .h format.

The next version of Grbl_ESP32 eliminates the need for the machine definition files. Unless you are doing custom coding, everyone will use the same compiled firmware. Configurations are done interactively or uploaded.

closed 2021-02-20 by bdring