bdring/Grbl_Esp32 #775: WebUI, files on not shown

JensHauser, 2021-02-17:
Please only submit bugs for **latest** main or devt branches. You can check the version number in the startup messgaes and compare it to the version in [grbl.h](https://github.com/bdring/Grbl_Esp32/blob/main/Grbl_Esp32/src/Grbl.h)

Please answer the following questions.

What version of the firmware are you using?
Most current as of GitHub main

Is the problem repeatable?
yes

Under what conditions does the bug occur?
- Click at any file, name or blue symbol, to show file content on the screen. E.g. subfolder "/Positionierung", file "goUp.nc"
- A new browser tab show up, URL e.g. "http://192.168.x.x/SD/Positionierung/goUp.nc"
- An error message comes up **cannot find /Positionierung/goUp.nc**

luc-github, 2021-02-17:
indeed it is not working when is was before - I am checking what happened

luc-github, 2021-02-17:
@JensHauser  please give a try to https://github.com/bdring/Grbl_Esp32/pull/776

It is working now per my tests

JensHauser, 2021-02-20:
Works in Devt, thanks, Luc and Mitch

JensHauser, 2021-02-25:
Taken over in main branch

closed 2021-02-25 by JensHauser