bdring/Grbl_Esp32 #778: incorrect mcu direction

sixtwoeight, 2021-02-18:
last week i had my little nephew over to visit (he's not quite so little anymore) i was showing him what one can do with cnc'ing at home and the board and all... i let him take the esp board off and get a good look at it. i wasn't pay ing close enough attention when i let him put it back and he put it facing the wrong direction (with the usb connector at the far end). powered it up and realised it was wrong when nothing worked. so i powered it down. my assumption is that it probably hurt the mcu more than anything so i've already ordered a replacement. but before i power it up again i'd like to know if there's a chance that it damaged the 6 axis board at all.

thanks in advance

bdring, 2021-02-19:
I have done that. The the board survived in my case. 

Don't reverse a motor driver. That kills just about everything.

joedirium, 2021-02-19:
I can confirm this, as I initially installed the MCU in reverse direction, which did not harm any of the boards. 

Anyway, I always keep spare MCU boards for my projects as they are pretty cheap if shipped from China.
I even keep a spare 6-pack board for the opposite reason because shipping from the US is pretty expensive and customs process is time consuming and annoying.

closed 2021-02-19 by bdring