bdring/Grbl_Esp32 #783: Problems Compiling Firmware - cannot convert 'esp_interface_t' to 'wifi_interface_t'

Protoneer, 2021-02-26:
I am following the wiki on how to compile GRBL-ESP32(Latest main branch) in Arduino 1.8.13. 

No changes made to the code.

This is the error I am getting from the Arduino IDE:
`
C:\Users\IT\AppData\Local\Temp\arduino_build_292218\sketch\src\WebUI\WifiConfig.cpp: In static member function 'static const char* WebUI::WiFiConfig::info()':
C:\Users\IT\AppData\Local\Temp\arduino_build_292218\sketch\src\WebUI\WifiConfig.cpp:74:54: error: cannot convert 'esp_interface_t' to 'wifi_interface_t' for argument '1' to 'esp_err_t esp_wifi_get_config(wifi_interface_t, wifi_config_t*)'
             esp_wifi_get_config(ESP_IF_WIFI_AP, &conf);
                                                      ^
C:\Users\IT\AppData\Local\Temp\arduino_build_292218\sketch\src\WebUI\WebSettings.cpp: In function 'Error WebUI::showSysStats(char*, WebUI::AuthenticationLevel)':
C:\Users\IT\AppData\Local\Temp\arduino_build_292218\sketch\src\WebUI\WebSettings.cpp:469:68: error: cannot convert 'esp_interface_t' to 'wifi_interface_t' for argument '1' to 'esp_err_t esp_wifi_get_protocol(wifi_interface_t, uint8_t*)'
                     esp_wifi_get_protocol(ESP_IF_WIFI_STA, &PhyMode);
                                                                    ^
C:\Users\IT\AppData\Local\Temp\arduino_build_292218\sketch\src\WebUI\WebSettings.cpp:504:58: error: cannot convert 'esp_interface_t' to 'wifi_interface_t' for argument '1' to 'esp_err_t esp_wifi_get_config(wifi_interface_t, wifi_config_t*)'
                 esp_wifi_get_config(ESP_IF_WIFI_AP, &conf);
                                                          ^
Multiple libraries were found for "SD.h"
 Used: C:\Users\IT\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.5\libraries\SD
 Not used: C:\Program Files (x86)\Arduino\libraries\SD
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\IT\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.5\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
`

I was able to compile this on my old PC but have done a fresh OS install and getting this issue.

For some reason I recall having the same issue but I can't seem to find what I did back then to solve it...

luc-github, 2021-02-26:
it is fixed here https://github.com/bdring/Grbl_Esp32/pull/782

Protoneer, 2021-02-26:
> it is fixed here #782
Wow that was quick. Was that a know issue?

Do will this be accepted for the main branch?

luc-github, 2021-02-26:
I have fixed in ESP3D/ESP3DLib before with ESP32 git version, so I know the root cause already - I was in process for GRBL_ESP32 when @bdring raised me the issue yesterday, so I have pushed PR this morning for GRBL_ESP32,  actually it is my fault I am late to apply it for GRBL_ESP32 because I was busy 
Sorry

Protoneer, 2021-02-26:
No sorry needed...

Thanks for the fix. :)

Protoneer, 2021-02-27:
Tried the latest code and it work!! Thanks

closed 2021-02-27 by Protoneer