bdring/Grbl_Esp32 #786: Problems Compiling Firmware

neves1989, 2021-02-26:
Please answer the following questions:

Have you read the [wiki regarding how to compile](https://github.com/bdring/Grbl_Esp32/wiki/Compiling-the-firmware)?
yes
What version of the Arduino IDE are you using? 
1.8.13
What version (commit date) of the [Arduino core for the ESP32](https://github.com/espressif/arduino-esp32) are you using? 
latest
Are you using the master branch of Grbl_ESP32?

Have you made any edits or configuration changes (list them) to the  firmware?

im trying to use unipolar motors and when i go to machine.h file and uncomment the atari machine it gives me the following error
Please paste the compiler error text here:

sketch/src/CustomCode.cpp:7:34: fatal error: Custom/atari_1020.cpp: No such file or directory
compilation terminated.
Multiple libraries were found for "WiFi.h"
 Used: /home/pi/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi
 Not used: /home/pi/Desktop/arduino-1.8.13/libraries/WiFi
Multiple libraries were found for "SD.h"
 Used: /home/pi/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/SD
 Not used: /home/pi/Desktop/arduino-1.8.13/libraries/SD
exit status 1
Error compiling for board ESP32 Dev Module.

bdring, 2021-02-26:
I think you might need to use PlatformIO to compile anything with files in the custom folder. You probably are not intending to use use that file and could comment it out. It contains special functions for the Atari printer.

neves1989, 2021-02-26:
Ok then thank you for your  help, i will try it tomorrow and check if works, i never used platformio before.

I was trying to load the atari machine because i have a small plotter (and your atari is the running with unipolar motors just what i nede) with grbl arduino uno and i would like to migrate it to esp32 because of its wireless capabilities but i only found examples for nema motors and my ploter runs 3 5v steppers with 3 uln2003 drivers.

Sorry for my english thank you for your help  e will try it tomorrow

neves1989, 2021-03-06:
I have the code compiled with platformio you were right thanks. I was testing one of the motors and it wont move the leds on the uln2003 driver blink when i send a jog command but the motor dont move should i need a 3v3 to 5v logic comverter?

Walkerlkr, 2023-08-14:
> 
你好，我也在使用uln2003 驱动器进行控制，你的问题解决了没