bdring/Grbl_Esp32 #788: Can't get out of alarm mode.

Protoneer, 2021-02-27:
I am testing my end stops and I am getting stuck in alarm mode.
Using the latest code from Main.
Switch on $21=1 and it happens....

```
<Idle|MPos:0.000,0.000,0.000,0.000,0.000,0.000|FS:0,0>
ALARM:1
<Alarm|MPos:0.000,0.000,0.000,0.000,0.000,0.000|FS:0,0>

```

No indication of the pin that is causing the issue.

When I manually trigger an end stop I get the following.. (GND to one of the end stop pins)

```
<Alarm|MPos:0.000,0.000,0.000,0.000,0.000,0.000|FS:0,0|Pn:X>
```

Hardware:
I have 10K pull-ups on all of the End stop pins. No 100nF Caps yet.

Machine settings:
```C++
#define MACHINE_NAME            "6 Pack Controller V1 (StepStick)"
#define N_AXIS 6

// I2S (steppers & other output-only pins)
#define USE_I2S_OUT
#define USE_I2S_STEPS
//#define DEFAULT_STEPPER ST_I2S_STATIC

#define USE_STEPSTICK   // makes sure MS1,2,3 !reset and !sleep are set

#define I2S_OUT_BCK             GPIO_NUM_22
#define I2S_OUT_WS              GPIO_NUM_17
#define I2S_OUT_DATA            GPIO_NUM_21

// Motor Socket #1
#define X_DISABLE_PIN           I2SO(0)
#define X_DIRECTION_PIN         I2SO(1)
#define X_STEP_PIN              I2SO(2)
#define X_STEPPER_MS3           I2SO(3)

// Motor Socket #2
#define Y_DIRECTION_PIN         I2SO(4)
#define Y_STEP_PIN              I2SO(5)
#define Y_STEPPER_MS3           I2SO(6)
#define Y_DISABLE_PIN           I2SO(7)

// Motor Socket #3
#define Z_DISABLE_PIN           I2SO(8)
#define Z_DIRECTION_PIN         I2SO(9)
#define Z_STEP_PIN              I2SO(10)
#define Z_STEPPER_MS3           I2SO(11)

// Motor Socket #4
#define A_DIRECTION_PIN         I2SO(12)
#define A_STEP_PIN              I2SO(13)
#define A_STEPPER_MS3           I2SO(14) 
#define A_DISABLE_PIN           I2SO(15)

// Motor Socket #5
#define B_DISABLE_PIN           I2SO(16)
#define B_DIRECTION_PIN         I2SO(17)
#define B_STEP_PIN              I2SO(18)
#define B_STEPPER_MS3           I2SO(19) 

// Motor Socket #5
#define C_DIRECTION_PIN         I2SO(20)
#define C_STEP_PIN              I2SO(21)
#define C_STEPPER_MS3           I2SO(22)
#define C_DISABLE_PIN           I2SO(23)

#define X_LIMIT_PIN             GPIO_NUM_33  // Sg1
#define Y_LIMIT_PIN             GPIO_NUM_32  // Sg2
#define Z_LIMIT_PIN             GPIO_NUM_35  // Sg3
#define A_LIMIT_PIN             GPIO_NUM_34  // Sg4
#define B_LIMIT_PIN             GPIO_NUM_39  // Sg5
#define C_LIMIT_PIN             GPIO_NUM_36  // Sg6

#define SPINDLE_TYPE            SpindleType::PWM
#define SPINDLE_OUTPUT_PIN      GPIO_NUM_26
#define SPINDLE_ENABLE_PIN      GPIO_NUM_4
#define SPINDLE_DIR_PIN         GPIO_NUM_16
#define COOLANT_MIST_PIN        GPIO_NUM_27

#define USER_DIGITAL_PIN_0      GPIO_NUM_27 //  M62 M63
#define USER_DIGITAL_PIN_1      GPIO_NUM_25 //  M62 M63
#define USER_DIGITAL_PIN_2      GPIO_NUM_27 //  M62 M63

#define CONTROL_SAFETY_DOOR_PIN GPIO_NUM_15  // labeled Door,  needs external pullup
#define CONTROL_RESET_PIN       GPIO_NUM_12  // labeled Reset, needs external pullup
#define CONTROL_FEED_HOLD_PIN   GPIO_NUM_0  // labeled Hold,  needs external pullup
#define CONTROL_CYCLE_START_PIN GPIO_NUM_14  // labeled Start, needs external pullup

#define DEFAULT_STEP_PULSE_MICROSECONDS I2S_OUT_USEC_PER_PULSE
```

Protoneer, 2021-02-27:
When I press $X nothing happens

Ctrl+ X  and then $X
```
<Alarm|MPos:0.000,0.000,0.000,0.000,0.000,0.000|FS:0,0>

Grbl 1.3a ['$' for help]
[MSG:'$H'|'$X' to unlock]
[MSG:Caution: Unlocked]
ok
ALARM:1
[MSG:Reset to continue]
```

bdring, 2021-02-27:
That is a hard limit alarm. An switch must be triggering the alarm. I would assume it is noise.

Join us on Discord to discuss further.

https://discord.gg/FPD8xkwxBk

Protoneer, 2021-02-28:
This happens as soon as I link Pin 36 and 39 to Endstop B and C

bdring, 2021-02-28:
Do you have external pullups? Those pins require external pullups.

Protoneer, 2021-02-28:
> Do you have external pullups? Those pins require external pullups.

Yes... I have a 10K on all of them. 

[Thanks for the code to show what pin is causing the issue.](https://github.com/bdring/Grbl_Esp32/commit/9ff2f4bc387361650b819a2c55649682cad9b827)