bdring/Grbl_Esp32 #789: espduino.h missing from the main branch

gkokolakis, 2021-03-02:
Hello,

I just bought and espduino32 board and i want to loaded it with grbl_esp32 in order to use it with a cnc shield v3. In the past i had seen a definition for this board in the machine's definition folder but it does not longer exist. Isn;t this board supported any more,

Regards, 

George

bdring, 2021-03-02:
We run automated tests on all machine definition files. We have to keep the file count down in the folder to keep the test duration to a reasonable amount.

 That file was removed, because it was not very popular. 

You will have to recreate the file yourself or search git history for it. I don't know how to do that, but a Google search can probably tell you.

bdring, 2021-03-02:
I looked at a very old branch and found it.

https://github.com/bdring/Grbl_Esp32/tree/tapster_pro/Grbl_Esp32/src/Machines

gkokolakis, 2021-03-02:
Thank you very much!

George

closed 2021-03-02 by bdring