bdring/Grbl_Esp32 #792: Firmware not working on wemos d1 r32 board

atrejo, 2021-03-05:
Please only submit bugs for **latest** main or devt branches. You can check the version number in the startup messgaes and compare it to the version in [grbl.h](https://github.com/bdring/Grbl_Esp32/blob/main/Grbl_Esp32/src/Grbl.h)

Please answer the following questions.

What version of the firmware are you using?
Latest
Is the problem repeatable?
Yes
Under what conditions does the bug occur?
Burn the grbl_esp32 to wemos d1 r32 board and after that the boards loops restarting
**Important** If you paste firmware code, please use [Markdown Code and Syntax Highlighting](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code) with language C++. Use the three back tick method.

```C++
  #define EASIER_TO_READ true
```

atrejo, 2021-03-05:
After burning the code to this board
Wemos D1 R32 ESP32
https://a.aliexpress.com/_mNxUCKf
And connect throw his or arduino ide it shows that the board is just loop restarting
[image]

gkokolakis, 2021-03-05:
Hello,

A few days ago i used a similar board. Are you having a cnc shield attached? If yes can you try removing it and report what happend? 
I solved the problem by following the instructions here: https://github.com/bdring/Grbl_Esp32/issues/314

Regards,

George, SV3QUP

atrejo, 2021-03-05:
Nop i don't have anything connected just the usb cable, same result with the typical ESP32 board

bdring, 2021-03-05:
@atrejo Are you able to load examples programs? Here is a very simple one.

[image]

atrejo, 2021-03-05:
Yes I can load them but after any program loads it starts rebooting, I loaded the example reboot reason and it says deep sleep

atrejo, 2021-03-05:
To me this ESP32 is a buggy and problematic board, I never faced this kind of problem with any other controller, arduinos, black pill, attiny, soon I'll get the seeeduino xiao which I expect will be a great replacement for this boards

closed 2021-03-05 by bdring