bdring/Grbl_Esp32 #802: Unable to interact with SD card via web UI

bcongdon, 2021-03-07:
- What version of the firmware are you using?
  - be08690827387aba086100a0cfedf3f763cff446
- Is the problem repeatable?
  - Yes.
- Under what conditions does the bug occur?
  - Any interaction with the web UI that uses the SD card (refreshing the contents, trying to upload a file, trying to create a directory, etc.) causes the ESP32 to reset (e.g. turn off completely and reboots).
    - I'm not sure if this is just a result of the ESP32 reset, but some of the SD card interactions cause my Z-axis servo to move as well
  - This only applies to using the web UI. I can interact with the SD card using the serial interface (e.g. `$SD/List`, `$SD/Run=...`) without any issue.

Looking at the patch nodes for be08690827387aba086100a0cfedf3f763cff446, it looks like some of the interactions with the SD card have changed recently. I wonder if this could be part of the issue?

I know this isn't much to work off of, so please let me know if there's a way I could provide better logs or more actionable details. Thanks!

lkarlslund, 2021-05-27:
Possibly the same problem as this one: https://github.com/bdring/Grbl_Esp32/issues/841

bdring, 2021-05-27:
The boot messages are very helpful

https://github.com/bdring/Grbl_Esp32/wiki/Requesting-Help#boot-messages

lkarlslund, 2021-05-28:
Here's a dump of a crash. Firmware is compiled with Arduino ESP 1.06, and the problem goes away with a lower version (ESP 1.05 works fine).

Reproduction is just pressing "refresh" on the SD card in the WEB UI.

`[MSG:Grbl_ESP32 Ver 1.3a Date 20210424]
[MSG:Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:Using machine:6 Pack External XYYZAB LAK]
[MSG:Axis count 5]
[MSG:I2S Steps, Stream]
[MSG:Door switch on pin GPIO(36)]
[MSG:Hold switch on pin GPIO(25)]
[MSG:Start switch on pin GPIO(39)]
[MSG:User Digital Output:0 on Pin:GPIO(15)]
[MSG:User Analog Output:0 on Pin:GPIO(12) Freq:5000Hz]
[MSG:Init Motors]
[MSG:X  Axis Standard Stepper Step:I2SO(2) Dir:I2SO(1) Disable:I2SO(0) Limits(0.000,1242.000)]
[MSG:Y  Axis Standard Stepper Step:I2SO(5) Dir:I2SO(4) Disable:I2SO(7) Limits(-1480.000,0.000)]
[MSG:Y2 Axis Standard Stepper Step:I2SO(10) Dir:I2SO(9) Disable:I2SO(8) Limits(-1480.000,0.000)]
[MSG:Z  Axis Standard Stepper Step:I2SO(13) Dir:I2SO(12) Disable:I2SO(15) Limits(-127.000,0.000)]
[MSG:A  Axis Standard Stepper Step:I2SO(18) Dir:I2SO(17) Disable:I2SO(16) Limits(-300.000,0.000)]
[MSG:B  Axis Standard Stepper Step:I2SO(21) Dir:I2SO(20) Disable:I2SO(23) Limits(-300.000,0.000)]
[MSG:Initializing RS485 VFD spindle]
[MSG:VFD RS485  Tx:GPIO(26) Rx:GPIO(16) RTS:GPIO(4)]

[MSG:Client Started]
[MSG:Connecting MySSID]
[MSG:Connecting.]
[MSG:VFD: VFD settings read: RPM Range(6000, 24000)]]
[MSG:Connecting..]
[MSG:Connecting...]
[MSG:Connecting....]
[MSG:Connecting.]
[MSG:Connecting..]
[MSG:Connecting...]
[MSG:Connecting....]
[MSG:Connected with 1.1.0.1]
[MSG:Start mDNS with hostname:http://LaHenrique.local/]
[MSG:SSDP Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]
[MSG:Flood coolant on pin GPIO(13)]
[MSG:Mist coolant on pin GPIO(14)]
[MSG:X  Axis limit switch on pin GPIO(33)]
[MSG:Y  Axis limit switch on pin GPIO(32)]
[MSG:Y2 Axis limit switch on pin GPIO(35)]
[MSG:Z  Axis limit switch on pin GPIO(34)]
[MSG:Probe on pin GPIO(2)]

Grbl 1.3a ['$' for help]
[MSG:'$H'|'$X' to unlock]
***ERROR*** A stack overflow in task clientCheckTask has been detected.
abort() was called at PC 0x40090bf0 on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x4009095c:0x3ffd41d0 0x40090bd9:0x3ffd41f0 0x40090bf0:0x3ffd4210 0x400934d7:0x3ffd4230 0x400950a0:0x3ffd4250 0x40095056:0xa5a5a5a5

Rebooting...`

It then continuously boot and crashes with the same messages for a while, until the browser gives up, and then the firmware survives.

Here's another backtrace from the same crash loop:

`Backtrace: 0x4009095c:0x3ffbb6c0 0x40090bd9:0x3ffbb6e0 0x40090bf0:0x3ffbb700 0x400934d7:0x3ffbb720 0x400950a0:0x3ffbb740 0x40095056:0x00000000`

Hope this helps.

@bdring thanks for the 6-axis board - I got my machine converted last weekend, and everything is looking fine so far. Great job on the PCBs!

bdring, 2021-05-28:
Can you attach your machine definition file. I can try to run it.

The "A stack overflow in task clientCheckTask has been detected." is a good clue. The task might be using too much memory.

bdring, 2021-06-03:
You must compile with ESP32 library version 1.0.3 in the Arduino IDE or via PlatformIO

closed 2021-06-03 by bdring