bdring/Grbl_Esp32 #805: Tips on debugging/verbose output (Machine locks up and drives the Z down when running macros)

jelsdon, 2021-03-08:
What version of the firmware are you using?

1.3a 20210203

Is the problem repeatable?

Yes - albeit intermittent (3/5 times)

Under what conditions does the bug occur?

Running macro - immediately following the completed execution of the first line. This is consistent and has not occurred
within a given macro run should the second line execute or prior to the first.

--

Hello grbl_esp32 community - I'm having unexpected behavior, could be firmware, could be config, could be wiring and mostly interested into what options do I have to get verbose/debugging information out of the controller; to see what it's thinking.

I've had a good run with Grbl_Esp32 until now when I've attempted using the probe, specifically when driven by a macro

When sending the gcode directly it has been successful, however once moving to macro I've found that
after the first line completes execution, the Z axis will start a very slow (~0.5mm/sec) controlled decline.

The machine becomes unresponsive at this point (Z continues to drop, wifi connection errs out)  and requires a power cycle - yet to see if any info is available via the serial console.

This has occurred both with the macro residing on ESP and SD.
I have not previously experienced this behavior when playing jobs off the SD but will retest now with the probe wired.

probe macro gcode available for completeness below, however have had this occur in macros containing just  X0 Y0 Z0 

```
G91 G38.2 Z-20 F200
G91 G1 Z4.000 F400
G4 P0.25
G91 G38.2 Z-5.0 F40
G4 P0.25
G10 L20 P0 Z17.46
G4 P0.25
G90 G54 G0 Z20 F400
```

Probe triggers a relay which breaks PROBE PIN's connection to ground (GPIO_NUM_4) -- with Probe/Invert on.

bdring, 2021-03-12:
I'll look at this when I get a chance.

What are you trying to do? Is this a macro to set work 0 using a probe?

bdring, 2021-03-12:
I can repeat the problem. It appears to only occur if it is done before any motion.

What hardware are you running on?

jelsdon, 2021-03-13:
Hi @bdring - I'm running Grbl_Esp32 on a ESP32-DevKitC-32UE

> What are you trying to do? Is this a macro to set work 0 using a probe?
Yep, the macro is to set work 0 (on Z) using a probe 

> I can repeat the problem. 
Good news! I hope no touch plates / end mills were harmed in the process

bdring, 2021-03-13:
I was asking about the controller hardware, not the ESP32. I want to know what step generation method is being used?

jelsdon, 2021-03-13:
For each of X Y Y' Z I have 
- DM524 Stepper driver
- 3A Nema23 425oz Stepper motor

Limit switches (normally closed) for homing terminate onto relays which trigger the esp32 (GPIO_NUM 15,16,17).
Probe is also connected to a relay which when triggered disconnects PROBE_PIN (GPIO_NUM_4) from ground. DEFAULT_INVERT_PROBE_PIN is set to 1.

(Full grbl_esp32 config can be found https://github.com/jelsdon/grbl-cncsys-config/commit/f10afe0e1a10f2f056329fd012e9545772decf5e )

readeral, 2021-09-29:
@bdring I've run into this problem today using the macro feature running a .g file from SPIFFS. Exactly the same issue, `G91 G38.2 Z-25 F100` doesn't progress to the next gcode line (which is to save  but rather locks the ESP3D interface up and the Z slowly lowers until an endstop is switched.
The full file contains these lines:
```
G21 G91
G38.2 Z-25 F100
G10 L20 P0 Z5
G0 Z3
```
Running each line manually through the Send Command window results in the correct outcome.

To answer the same question @jelsdon answered, I'm running DM556 Stepper drivers, 4A Nema23 425oz stepper motors. My  controller is https://github.com/RootCNC/ESP32-GRBL-Root-Controller by @RootCNC 

Has there been a fix for this found?

I have had other issues today after running G38.2 for all 3 axis manually, which may or may not be connected. I'll raise another issue for that.

dfict, 2024-12-30:
did anyone find a fix for this issue? even if it's a workaround? i have a legacy system running esp32 that i added a touchplate to, this is the same crash behavior i'm having. just trying to make a macro that works. manual entry works, macro running makes the same intermittent crash.

MitchBradley, 2024-12-30:
As far as I know nobody has done any updates to Grbl_Esp32 for a long time.  All of the development and maintenance effort is on FluidNC.  FluidNC has a lot of stability improvements in addition to many new features.