bdring/Grbl_Esp32 #812: Laser engraving speed

DevEvV, 2021-03-11:
When engraving a bitmap, the movement speed drops very dramatically. I increased the accelerations to the maximum, but it didn't help. Obviously the limiting factor is the serial port buffer. Is there any way to increase RX_BUFFER_SIZE to 1024 or even more?

MitchBradley, 2021-03-12:
Without knowing the GCode, the limiting factor cannot be determined.  It could be the serial line bandwidth, irrespective of the buffer size.

ZDVstore, 2021-03-12:
#define BLOCK_BUFFER_SIZE   513

For using it I should change code in "Planner.h" & "Planner.cpp"!

Planner.h:
uint16_t plan_next_block_index(uint16_t block_index);
uint16_t plan_get_block_buffer_available();
uint16_t plan_get_block_buffer_count();

Planner.cpp:
static plan_block_t block_buffer[BLOCK_BUFFER_SIZE];  // A ring buffer for motion instructions
static uint16_t      block_buffer_tail;                // Index of the block to process now
static uint16_t      block_buffer_head;                // Index of the next block to be pushed
static uint16_t      next_buffer_head;                 // Index of the next buffer head
static uint16_t      block_buffer_planned;             // Index of the optimally planned block
uint16_t plan_next_block_index(uint16_t block_index) {
static uint16_t plan_prev_block_index(uint16_t block_index) {
uint16_t plan_get_block_buffer_available() {
uint16_t plan_get_block_buffer_count() {

If you plan to use 1024, you will have to disable Bluetooth. Otherwise, there is not enough DRAM!!!

I suffered for 2 months and realized that the scheduler for ESP 32_gr bl works very bed, even with the WiFi turned off. I use the laser mode with 16-bit PWM. This is very important for my 2500 mW laser for getting a bitmap image.
At the same time, I always have problems hanging at speeds over 2000 and accelerations over 500, regardless of the BLOCK_BUFFER_SIZE size. If I disable WiFi, the problem repeats periodically...

I have been using the atmega328 chip and grbl firmware for a long time and had no problems with accelerations up to 1500 and speeds up to 4000 when engraving bitmaps.

I tried all the firmware (MAIN & DEV) on the date 11.03.2021.

ZDVstore, 2021-03-12:
I have been using the ESP 32 chip (Fysetc E4 v1.0) on my 3D printer for 2 months now. The printer runs on Marlin and works amazingly.

ZDVstore, 2021-03-12:
> Without knowing the GCode, the limiting factor cannot be determined. It could be the serial line bandwidth, irrespective of the buffer size.

I've done a lot of experiments with the exchange rate. I am sure that it is not the line, but the fact that the planner does not cope with the data flow!!!

You can see my gcode and logs:
https://github.com/bdring/Grbl_Esp32/issues/810

DevEvV, 2021-03-12:
> #define BLOCK_BUFFER_SIZE 513
> 
> For using it I should change code in "Planner.h" & "Planner.cpp"!

Приветствую, Дмитрий. Эту операцию я уже проделывал не особо помогло. Может я хочу от контроллера невозможного. У меня CO2 лазер и станок спокойно гоняет ось X со скоростью 60000 мм/s. Хотелось бы заставить его гравировать со скоростью 30000. Если гравировать однотонные крупные узоры, скорость почти не падает. Как только в программе появлется растер или дизеринг с большим количеством мелких перемещений, станок начинает плестись как черепаха. Не знаю что крутить, может он тупо не способен на такое. Нужно делать специальный мод под гравировку,  который будет отсылать все команды перемещения по X до момента пока не изменится координата Y, запускать одно движение по Х и в процессе этого движения просто менять мощность лазера в определенных точках, не высчитывая скорости и ускорения. Но я не программист и такое не осилю.

Что касается вашей проблемы. Говорят, эти контроллеры требовательны к питанию. У меня на длинном проводе возникают проблемы, возможно из-за падения напряжения. На коротком качественном проводе подключенном к порту USB 3.0 все работает стабильно на любых скоростях.

ZDVstore, 2021-03-12:
Я уже 2 месяца мучаюсь!!! Не хочет нормально планировщик работать и все тут - прошивка сырая!
Работа, конечно, очень сложная по переносу GRBL на 32 битную систему. Мне кажется, на данный момент разработчики не перенесли нормально код планировщика в ESP32 RTOS.
Максимум что удалось - сделать растровую картинку 100*75 на скорости 500 при ускорении 250 с SD карты. С USB порта 200*150, но видно, что при попытке двигаться выше 2000, планировщик не успевает и снижает скорость. Через раз виснет наглухо, даже с выключенным WiFi и BlueTooth!
В общем, никак не работоспособна прошивка в режиме лазера. 16 битный ШИМ сильно бы помог. От WiFi пользы никакой без стабильности основных функций. Будем ждать!

ZDVstore, 2021-03-12:
Питание у меня организовано от внешнего БП. Думаю, проблем здесь нет. Да и тестирую в основном без использования USB, с карты памяти.

ZDVstore, 2021-03-12:
На Marlin на плате Fysetc E4 v1.0 вообще проблем нет!!! Управляет моим CORE-XY принтером на скоростях печати до 200. Работает весь функционал. Печатаю с SD карты, поскольку на дисплей ног Fysetc не отвел.
Делаю вывод, что дело не в ESP32 - этот процессор вполне способен вытянуть гравер!

DevEvV, 2021-03-12:
> Питание у меня организовано от внешнего БП. Думаю, проблем здесь нет. Да и тестирую в основном без использования USB, с карты памяти.

WiFi там вообще глючный. Пробовал запускать через Telnet вылетает. Там у них какая-то проблема с переполнением буфера они вроде знают, но пока не решают. А через USB у меня все отлично! Никаких глюков на ваших скоростях не должно быть! Возможно там дело не в скоростях, а в количестве мелких команд перемещения.

ZDVstore, 2021-03-12:
Именно! Если по вектору двигаться - не вопрос, жгет на любых скоростях. Как только раст с диапазоном изменения ШИМ от 19500 до 55700 виснет и все на этом...

ZDVstore, 2021-03-12:
Кстати, при работе от USB хорошо видно, что станок начинает сильно подтормаживать. А я сразу на диагонали пробую растр гравировать - тут мелких перемещений множество. К примеру, если расчетное время работы от LaserGRBL около 30 минут, то станок пробегает по GCOD где-то за 50 минут.

DevEvV, 2021-03-12:
У меня, кстати, UniversalGcodeSender команды быстрее шлёт чем LaserGRBL. И это еще больше запутывает - не знаешь на что грешить.. 

Придется изучать Марлин или смотреть в сторону промышленных контроллеров..

bdring, 2021-03-12:
### Settle down everyone! 

Be nice or I will not participate in this issue. Grbl_ESP32 is a free and open source project with minimal sponsorship. Nobody is paid to support these questions.

@DevEvV I assume you are properly using laser mode and M4 command. That is essential for proper operation.

GCode was never designed for things like lasers where you have so many tiny moves each with spindle speed change. The planner is designed to help with smooth operation during changes in direction. This is not needed in laser engrave jobs and almost gets in the way.

Increasing the size of the planner buffer could make things worse (I don't know) because Grbl is re-planning a greater number of moves.

If you are using USB, you should consider raising the speed from 115200 baud to the highest your sender supports.

ZDVstore, 2021-03-12:
I am very grateful for your work and understand its complexity very well!!! I will monitor the project and test all updates... Thank you for your work!

ZDVstore, 2021-03-12:
> У меня, кстати, UniversalGcodeSender команды быстрее шлёт чем LaserGRBL. И это еще больше запутывает - не знаешь на что грешить..
> 
> Придется изучать Марлин или смотреть в сторону промышленных контроллеров..

Я оставил резервный вариант: MKS DLC v2.0 + MKS TFT24 Display (есть прошивка для работы с гравером) + ESP8266. Со следующей недели буду тестировать...

ZDVstore, 2021-03-16:
I have tested DEVT811 firmware. And for the first time I got a good result. I managed to print a 255x150 photo. The image was printed at a speed of 1250, acceleration of 2000, PWM resolution of 16 bits (MAX laser Power: 65535). It's a great job! Thank you!

[image]

It is my main configuration:

```
#define DEFAULT_STEP_PULSE_MICROSECONDS 5
#define DEFAULT_JUNCTION_DEVIATION            0.05
#define DEFAULT_X_ACCELERATION                    2000.0 // $120 - ОСЬ X
#define DEFAULT_Y_ACCELERATION                    2000.0 // $121 - ОСЬ Y

// #define USE_RMT_STEPS

#define RX_BUFFER_SIZE 255
#define TX_BUFFER_SIZE 255

#define BLOCK_BUFFER_SIZE      16    
#define SEGMENT_BUFFER_SIZE 16
#define LINE_BUFFER_SIZE          80

#define SPINDLE_TYPE SpindleType::LASER
#define DEFAULT_LASER_FULL_POWER    65536
#define DEFAULT_SPINDLE_RPM_MAX     65535
#define DEFAULT_SPINDLE_FREQ             1220 
#define DEFAULT_SPINDLE_MAX_VALUE  65535
#define SPINDLE_PWM_BASE_FREQ        1220
#define SPINDLE_PWM_BIT_PRECITION  16
#define SPINDLE_PWM_MAX_VALUE       65535
```

I have a question. It is possible to set the laser power limit 65535:
`laser_full_power = new IntSetting(EXTENDED, WG, NULL, "Laser/FullPower", DEFAULT_LASER_FULL_POWER, 0, 1000, checkSpindleChange);`
change to
`laser_full_power = new IntSetting(EXTENDED, WG, NULL, "Laser/FullPower", DEFAULT_LASER_FULL_POWER, 0, 65535, checkSpindleChange);`?

It works fine!

UniqueLemming, 2021-03-20:
I was having the same issue while trying to run engraving on my laser. A lot of stuttering even on simple engravings if I was over 3000mm/min. I changed the following in Config.h

#define BLOCK_BUFFER_SIZE 250
#define SEGMENT_BUFFER_SIZE 20
#define RX_BUFFER_SIZE 8192

This allowed me to go from max 3000mm/min engraving smoothly to as fast as my machine can travel. It's perfectly smooth now at max speed(currently 25000mm/min). No gittering at all. Please keep in mind that I don't really know what I'm doing. I copied these numbers from the GRBL-LPC firmware that runs on my Smoothieboard on my other machine. I did turn off Bluetooth, WiFi, and SD card before testing this just in case the ESP32 couldn't handle it all. I'll probably turn them back on at some point and see if it's still working okay. But since I only  connect via USB anyway, I might not get around to that right away.

solawc, 2021-03-25:
> I was having the same issue while trying to run engraving on my laser. A lot of stuttering even on simple engravings if I was over 3000mm/min. I changed the following in Config.h
>
> #define [...]

Hello, can you tell me if it is normal to print at 3000r / min after Bluetooth, WiFi and SD are turned on? I also met the same problem as you. I hope to discuss and solve it with you

MitchBradley, 2021-03-25:
The speed that you can achieve depends on the machine.  On my inexpensive laser engraver, I have to engrave at about 500 mm/min for good quality.

solawc, 2021-03-25:
> The speed that you can achieve depends on the machine. On my inexpensive laser engraver, I have to engrave at about 500 mm/min for good quality.

Interestingly, when I use tmc2209 and 4988 drivers for laser carving, the carving effect is very unsatisfactory. There are many dense breakpoints on a straight line, and serrations on a vertical line. This makes me very confused, and I have not solved it yet.But when I use MKS dlc2.0 motherboard and grbl firmware with almost the same parameters, the performance of DLC is very good.

truglodite, 2021-03-30:
I can confirm that 
```
#define BLOCK_BUFFER_SIZE 250
#define SEGMENT_BUFFER_SIZE 20
#define RX_BUFFER_SIZE 8192
```
makes my laser like a rocket ship via usb or sdcard now. 500mm/min is a bit slow for some materials. For example my cheap 3.5W laser etches fairly dark on MDF at 80% 50mm/sec (3000mm/min).

@solawc, it is hard to guage what could be wrong with the description, "many dense breakpoints on a straight line, and serrations on a vertical line". Based on your experiences with the MKS board, I am guessing your gcode process is ok. My guess is you don't have laser mode enabled in firmware. You have to use `#define SPINDLE_TYPE::LASER` etc for $32=1 and M4 to work right.

truglodite, 2021-04-04:
Reporting back in that the settings I posted directly above appear to be unstable. I have little proof that the modifications are causing crashes due to low memory, but it is acting like it is. 

I had the esp32 booted up for 24hrs without issues, only verified by being able to access the webui and serial port without issues. However it seemed about half the time I would go to do a level probing operation from bCNC for PCB milling, the controller would randomly crash right after probing a point. The leveling script in bCNC basically uses G0 to move around a grid of points, and does a G38.2 along z at each point. The crashes always appeared right after sending the probe data to bcnc (probe touching the PCB).

I should also add that, despite having to hard reset to recover from this crash during probing a few times, if the probing did finish up, I was always able to complete several cutting operations afterwards with no issues. So not 100% sure if there may be some other unrelated bug in the code that is triggered by doing multiple probing operations in a row, or if this is a side effect of low memory from my changes.

To experiment with this, I noted in the webui with the above settings, free RAM enumerated to 112kB. I reduced the modifications as follows:

```
#define BLOCK_BUFFER_SIZE 128       //250 or 16, default=15 or 16
#define SEGMENT_BUFFER_SIZE 16       //20 or 16, default=6
#define RX_BUFFER_SIZE 1024       //8192 or 255, default=256
```

Also, after getting more info dealing with issue #848, I changed from board manager version 1.0.4 to 1.0.5 for this test. So far I have not had any issues with probing. I will report back after I have had some time without issues. I sure hope this ends up working out, and that it is just an unrelated issue that needs fixing. If that is the case, this would make grblesp32 far more attractive to the growing group of users who are in search of improved performance.

bdring, 2021-04-04:
Not very many senders will take advantage of a bigger **RX_BUFFER_SIZE**. Many don't utilize the Rx buffer or they use an assumed fixed value of 128 or 256.

MitchBradley, 2021-04-04:
The dynamic buffer size reporting that is enabled by $Report/Status=3 ($10=3) is currently completely broken - the buffer length number that it reports in the Bf: field is just wrong.  Senders that use that feature are likely to be completely confused.

truglodite, 2021-04-05:
I just commented out the `RX_BUFFER_SIZE` part and got 4 good level probe ops in a row. Went back up to `BLOCK_BUFFER_SIZE 128` and `SEGMENT_BUFFER_SIZE 16`, and get the same successful result.  This is with $10=3 btw, which is showing `Bf: 127,16` at times, and `Bf: 64,0` other times. The buffer indicator in bCNC does not fill properly either.

truglodite, 2021-04-11:
My testing with these changes turned out to be a success:
```
#define BLOCK_BUFFER_SIZE 250          //250 or 16, default=15 or 16
#define SEGMENT_BUFFER_SIZE 16       //20 or 16, default=6
``` 
...no changes to `RX_BUFFER_SIZE`. To complete the picture, I have BT and SDcard disabled, but wifi, ota, webui enabled on my autosquaring mpcnc (6-pack board). I have done several projects on the machine during this test, including several 2-sided circuit boards, a few items out of hardwood, and some laser etches all without issues. That includes many surface level probings that I had issues with on my last test. 

The laser etching was tested using a 100mm/sec dithered image that was causing major stutter before these changes (would max out around 20mm/sec in practice). With these changes that file was averaging 5800mm/min in the middle of the raster lines. It looks good, and sounds smooth... I'd say 98% stutter free. My 2% downgrade is due to the fact that it I was observing motion that was asymmetrical. For example, on some areas of the image it was maxing out about 5800 going +Y, and 5400 going -Y... the -Y feed rate sounded like it was wandering slightly at those times. I can only describe that as a minor lingering stutter. Not sure if more tweaking could eliminate that.

For me, and I think most folks using diode lasers, the changes as is make this firmware 100% useable for multi-tool mills (laser+router+etc). My laser needs to stay around 50mm/sec or slower, depending on material, to get a good dark etch. At those speeds my machine runs a solid 3000mm/min... smooth as silk with no stutter.

solawc, 2021-04-20:
> I can confirm that
> 
> ```
> #define BLOCK_BUFFER_SIZE 250
> #define SEGMENT_BUFFER_SIZE 20
> #define RX_BUFFER_SIZE 8192
> ```
> 
> makes my laser like a rocket ship via usb or sdcard now. 500mm/min is a bit slow for some materials. For example my cheap 3.5W laser etches fairly dark on MDF at 80% 50mm/sec (3000mm/min). I have also tried with wifi enabled and it is fine.
> 
> @solawc, it is hard to guage what could be wrong with the description, "many dense breakpoints on a straight line, and serrations on a vertical line". Based on your experiences with the MKS board, I am guessing your gcode process is ok. My guess is you don't have laser mode enabled in firmware. You have to use `#define SPINDLE_TYPE::LASER` etc for $32=1 and M4 to work right.

No, I've been using this model and haven't changed it