bdring/Grbl_Esp32 #813: Error homing Laser Grbl Esp32

RobertoRmz, 2021-03-12:
I am mounting a CNC Laser with the 38-pin ESP32 and I am using the GRBL Laser Software, **firmware bdring / Grbl_Esp32**, after several days I was able to make the necessary configurations to move the nema 17 stepper motors and mount a 5W laser to it. 

It worked very well. But when setting the Limit Switches so that it can do Homing, the following error appears, **Homing Fail** with the following message [**MSG: Cannot multi-axis home with squared axes. Homing normally**], I don't know how to remove this. 

[image]

The other problem I saw is that when I did homing, the Z axis appears in the corner and it is the first thing that moves, I don't know how to eliminate it if someone could give me some advice, I would greatly appreciate it. 

[image]

I was investigating how to eliminate the Z axis for a laser and I saw that there are these lines in config.h in a previous version of GRBL ESP32, in the existing version that is to download these lines do not appear, again I would greatly appreciate some advice on how to be able to eliminate the z axis to be able to do a homing correctly. 

This is the part of the code to remove the z-axis in GRBL ESP32 from the previous version in which it does not appear in the version you download: 

// NOTE: Defaults are set for a traditional 3-axis CNC machine. Z-axis first to clear, followed by X & Y.  
#define HOMING_CYCLE_0 (1<<Z_AXIS)	// TYPICALLY REQUIRED: First move Z to clear workspace.
#define HOMING_CYCLE_1 (1<<X_AXIS)  
#define HOMING_CYCLE_2 (1<<Y_AXIS)  

// NOTE: The following is for for homingg X and Y at the same time
// #define HOMING_CYCLE_0 (1<<Z_AXIS) // first home z by itself
// #define HOMING_CYCLE_1 ((1<<X_AXIS)|(1<<Y_AXIS))  // Homes both X-Y in one cycle. NOT COMPATIBLE WITH COREXY!!!

I would appreciate any comments because I have not found a solution to remove the z axis in the GRBL laser software, regards.

ZDVstore, 2021-03-12:
You can use only 2 axises:
#define N_AXIS 2

bdring, 2021-03-12:
Not a bug. Just correct your homing cycles.

We do not recommend using N_AXIS less than 3. some gcode senders will not like this and we have not tested the firmware with 2.

If you have N_AXIS as 3, but only define pins for X and Y, Grbl_ESP32 will define a virtual Z axis. You can still move a virtual axis and see the status, but no I/O is used. You cannot home it, because there is no motion or switch to hit.

To home X and Y, just define the homing cycles.
```cpp
#define HOMING_CYCLE_0 (1<<X_AXIS) 
#define HOMING_CYCLE_1 (1<<Y_AXIS)
```

You can define them live, without recompile by sending these commands.

$Homing/Cycle0=X
$Homing/Cycle1=Y
$Homing/Cycle2=
$Homing/Cycle3=
$Homing/Cycle4=
$Homing/Cycle5=

ZDVstore, 2021-03-12:
> Not a bug. Just correct your homing cycles.
> 
> We do not recommend using N_AXIS less than 3. some gcode senders will not like this and we have not tested the firmware with 2.
> 
> If you have N_AXIS as 3, but only define pins for X and Y, Grbl_ESP32 will define a virtual Z axis. You can still move a virtual axis and see the status, but no I/O is used. You cannot home it, because there is no motion or switch to hit.
> 
> To home X and Y, just define the homing cycles.
> 
> ```c++
> #define HOMING_CYCLE_0 (1<<X_AXIS) 
> #define HOMING_CYCLE_1 (1<<Y_AXIS)
> ```
> 
> You can define them live, without recompile by sending these commands.
> 
> $Homing/Cycle0=X
> $Homing/Cycle1=Y
> $Homing/Cycle2=
> $Homing/Cycle3=
> $Homing/Cycle4=
> $Homing/Cycle5=

Thanx!

RobertoRmz, 2021-03-13:
**Bdring** and **ZDVstore** thank you for taking the time to answer my question, I had been stuck with this problem for several days.

ZDV store try #define N_AXIS 2 but the same thing keeps happening, homing the Z in Laser GRBL The solution I found for it to do homing was in Machines I was using mpcnc_laser_module_v1p2 and changing it to pen_laser.

The only bad thing is when you do Homing in the Laser GRBL the two motors do the homing at the same time and I did not know how to make it homing the x axis first and then the y axis.

Then copy into the pen_laser folder the codes that Bdring gave me and it works correctly.

Thank you very much for your comments Bdring and ZDVstore, regards.

bdring, 2021-03-13:
The defaults.h file has these definitions.

```cpp
#ifndef DEFAULT_HOMING_CYCLE_0
#    define DEFAULT_HOMING_CYCLE_0 bit(Z_AXIS)
#endif

#ifndef DEFAULT_HOMING_CYCLE_1
#    define DEFAULT_HOMING_CYCLE_1 (bit(X_AXIS) | bit(Y_AXIS))
#endif

#ifndef DEFAULT_HOMING_CYCLE_2
#    define DEFAULT_HOMING_CYCLE_2 0
#endif

#ifndef DEFAULT_HOMING_CYCLE_3
#    define DEFAULT_HOMING_CYCLE_3 0
#endif

#ifndef DEFAULT_HOMING_CYCLE_4
#    define DEFAULT_HOMING_CYCLE_4 0
#endif

#ifndef DEFAULT_HOMING_CYCLE_5
#    define DEFAULT_HOMING_CYCLE_5 0
#endif
```

If you put any in your machine definition file they will be used like...
```cpp
#define DEFAULT_HOMING_CYCLE_0 bit(X_AXIS)
#define DEFAULT_HOMING_CYCLE_0 bit(Y_AXIS)
```

There is info on the wiki as well

https://github.com/bdring/Grbl_Esp32/wiki/Setting-Up-Limit-Homing-Switch#homing-cycles

RobertoRmz, 2021-03-14:
Thank you very much **bdring** for your advice and for passing me the link of the limit switches I had not seen that information.

closed 2021-03-14 by bdring