bdring/Grbl_Esp32 #815: No check on 0 position

topogigio, 2021-03-14:
I don't know if is a bug, but is a problem.
Please answer the following questions.

What version of the firmware are you using? 
Last but I think that affect all version

Is the problem repeatable?
Yes

Under what conditions does the bug occur?
Every time you move each axes in endstop direction

During movment in endstop direction there is no check on 0 position so endstop goes on and hardware limit error occur.
Thanks

bdring, 2021-03-15:
Please explain further in detail.

What command did you send?
What happened?
What were you expecting to happen?

Do you have soft limits and hard limits enabled?

topogigio, 2021-03-17:
Hi,
I control grbl using esp3d.
I have hard limits enabled, after home command, If I go back in the endstop
direction using jog buttons, before activating the endstop my expectation
is that an error occurs "exceed...". but... click and I must reset the
board.
Max travel is calculated for each axis:
- make home axis;
- move at the end of the axis minus 1mm;
- read the value indicated in esp3d GUI;
- set the value in grbl field.

I made several tests with different values axis max travel, but I put the
last value for each axis, calculated using the previous method mentioned.
Thank you for share your job and for this email
Regards

Andrea Marotta
Amministratore unico
UPG S.r.l.s.

NOTA DI RISERVATEZZA
Il presente messaggio e-mail ed ogni file ad esso allegato contiene
informazioni di natura confidenziale e riservata, indirizzate
esclusivamente alla/e persona/e  individuata come destinatario. La lettura,
l’utilizzo, la riproduzione e la divulgazione a terzi, anche parziale, del
contenuto di questo messaggio è consentito soltanto al destinatario.La
diffusione, distribuzione e/o copiatura del documento trasmesso da parte di
qualsiasi soggetto diverso dal destinatario è proibita, sia ai sensi
dell’art. 616 c.p. sia ai sensi del D.Lgs. n. 196/2003. Chiunque ricevesse
questo messaggio per errore, è pregato di comunicare tempestivamente
l’errata trasmissione al mittente per e-mail all’indirizzo ***@***.***
e di cancellare il presente messaggio insieme agli eventuali allegati.
Grazie

CONFIDENTIALITY NOTICE
This e-mail and any file attached to it contain privileged and confidential
information to be transmitted only to the addressee.If you are not the
intended addressee, be advised that any review, use, copy or transmission
to a third party of all or part of the information contained in this email
is prohibited. If you have received this message in error, please notify
immediately the sender by email at ***@***.*** and delete this email
and its attachments from your system. Thank you.

bdring, 2021-03-17:
If you have hard limits enabled, the firmware will create a hard limit alarm 1 if you hit a limit switch. The machine will stop suddenly and there is a likely loss of position. You must reset and rehome.

If you have soft limits enabled and you request a move that will exceed the dimensions of the machine, the move will be rejected and a soft limit alarm 2 will be created. There is no loss in position, you must reset, but you do not need to rehome.

bdring, 2021-03-17:
https://github.com/bdring/Grbl_Esp32/wiki/Setting-Up-Limit-Homing-Switch#soft-limits