bdring/Grbl_Esp32 #818: Step and Dir with GPIO_NUM_25  and GPIO_NUM_26 Error (?)

nsiatras, 2021-03-15:
Hello,

I am trying to use GPIO_NUM_25 as Y_STEP_PIN and PIO_NUM_25 as Dir Pin
#define Y_STEP_PIN              GPIO_NUM_**25**
#define Y_DIRECTION_PIN         GPIO_NUM_**26**

When I define them as above then I cannot move the motor.
but... if I change the numbers to:
#define Y_STEP_PIN              GPIO_NUM_**26**
#define Y_DIRECTION_PIN         GPIO_NUM_**25**

then I can move the stepper motor but only to one direction. It never change direction..

Any advice ?

bdring, 2021-03-15:
It appears to me that `#define Y_STEP_PIN GPIO_NUM_26` is correct and `#define Y_DIRECTION_PIN GPIO_NUM_25` is not.

Are you sure the direction pin is connected to 25?

nsiatras, 2021-03-15:
Thanks for the reply.

My initial setup is :
#define Y_STEP_PIN GPIO_NUM_25
#define Y_DIRECTION_PIN GPIO_NUM_26
**This doesn't work at all**

If I change the pinout to:
#define Y_STEP_PIN GPIO_NUM_26
#define Y_DIRECTION_PIN GPIO_NUM_25
This moves the motor to one direction only.
Of course I make sure that I connect the pin **26** to the **Step** pin of the Driver and the Pin **25** to the **Dir** Pin of the Driver

nsiatras, 2021-03-15:
I just checked the GPIO_NUM_25 with my Oscilloscope...
It seems that it is at 3.3V (ON) all the time.
Do you think the dev board is damaged?

bdring, 2021-03-15:
Maybe, what CNC controller circuit (not ESP32) are you using? Is it a custom design or one you purchased?

The direction will will stay on in the direction of motion, even after it has stopped. It will only change if there is a change in direction. Make sure you measure the voltage after a forward move in positive Y and then after negative Y.

nsiatras, 2021-03-15:
I somehow managed to entirely damage the pin 25 on my Node MCU board
Anyway, after several hours trying to make it work I replaced the board
with an other one and it works fine.

Thank you for your help!

nsiatras, 2021-03-15:
PS. I was experimenting on a breadboard.

closed 2021-03-15 by nsiatras