bdring/Grbl_Esp32 #831: Absolute A Axis ?

nsiatras, 2021-03-18:
Is there any possible way to use A Axis without Soft limits ?
I want to make the A axis rotating at full 360 degrees.

bdring, 2021-03-18:
Not yet, in a future version we plan to allow all the homing and limit switch features to be applied per axis.

Do you use a home switch for that axis?

nsiatras, 2021-03-18:
I use limit switches only for X,Y and Z. 
I do not like limit switches and soft limits on rotary Axis like the A :)

nsiatras, 2021-03-18:
Is there any way I can extend the A axis max travel through a command I will send from the serial port?
I am currently building a 4 axis post processor for SprutCAM and a custom 4Axis GCode Sender.
Any advice would be very helpful

bdring, 2021-03-18:
You could set the max travel to a really large value and Home/Mpos to half of that. For example

$Z/MaxTravel = 10000
$Z/Home/Mpos=5000

will result in a range like this...(ignore everything but the limits)

[MSG:Z  Axis Trinamic TMC2130 Step:I2SO(10) Dir:I2SO(9) CS:I2SO(11) Disable:I2SO(8) Index:-1 R:0.110 Limits(-5000.000,5000.000)]

nsiatras, 2021-03-18:
I can change the work position with G92
If there was a way to change machine position with a GCode command I could make it work without outside the limits

bdring, 2021-03-18:
Yes, work position offset can be changed. I usually recommend [G10](http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g10-l20
) with [G54-G59](http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g54-g59.3) over G92.

MitchBradley, 2021-03-18:
I thing the question had to do with machine position, not work position.  There is a ticket on our Trello board, provisionally assigned to me, to add support for G28.2 and G28.3 for managing machine coordinates from GCode - but it is low on the priority list.

nsiatras, 2021-03-18:
@MitchBradley Yes. That I was asking for.

I work with Java, PHP and .NET....
Wish I could help you out with this project.

terjeio, 2021-03-18:
If max travel is set to 0 for an axis that can be used as input to not apply soft limits. I did that and should be easy to add to this port too.

nsiatras, 2021-03-19:
@terjeio I believe your solution is the simplest and best way to remove soft limits from any axis. If someone can guide me on where to look on the code I can make the change. Is it in the Limits.cpp ?

terjeio, 2021-03-19:
@nsiatras Here is how I do it, note that I check against -0 not 0. Homing and setting validation may need changes as well.

```
// Checks and reports if target array exceeds machine travel limits. Returns false if check failed.
// NOTE: max_travel is stored as negative
// TODO: only check homed axes?
bool system_check_travel_limits (float *target)
{
    bool failed = false;
    uint_fast8_t idx = N_AXIS;

    if(settings.homing.flags.force_set_origin) {
        do {
            idx--;
        // When homing forced set origin is enabled, soft limits checks need to account for directionality.
            failed = settings.axis[idx].max_travel < -0.0f &&
                      (bit_istrue(settings.homing.dir_mask.value, bit(idx))
                        ? (target[idx] < 0.0f || target[idx] > -settings.axis[idx].max_travel)
                        : (target[idx] > 0.0f || target[idx] < settings.axis[idx].max_travel));
        } while(!failed && idx);
    } else do {
        idx--;
        failed = settings.axis[idx].max_travel < -0.0f && (target[idx] > 0.0f || target[idx] < settings.axis[idx].max_travel);
    } while(!failed && idx);

    return !failed;
}
```

nsiatras, 2021-03-19:
I believe I made it.

I added the **&& axis_settings[idx]->max_travel->get()>0** case in the if statement of the **bool limitsCheckTravel** inside the **Limits.cpp** file.

I also set the A axis max travel to zero inside my Machine file
**#define DEFAULT_A_MAX_TRAVEL 			0 **
and I think it works good

```
// Checks and reports if target array exceeds machine travel limits.
// Return true if exceeding limits
bool limitsCheckTravel(float* target) {
    uint8_t idx;
    auto    n_axis = number_axis->get();
    for (idx = 0; idx < n_axis; idx++) {
        float max_mpos, min_mpos;
		
		if ((target[idx] < limitsMinPosition(idx) || target[idx] > limitsMaxPosition(idx)) 
			&& axis_settings[idx]->max_travel->get()>0) {
			return true;
		}
    }
    return false;
}
```

nsiatras, 2021-03-19:
Any ideas on how I can avoid the Error 80 when I change the Max Travel of A Axis to 0 (Zero) through serial command ?
$133 = 0

Should I like inside the Settings.cpp ?

nsiatras, 2021-03-19:
I changed the line to 
`auto setting = new FloatSetting(GRBL, WG, makeGrblName(axis, 130), makename(def->name, "MaxTravel"), def->max_travel, (axis>=3) ? 0 : 1.0, 100000.0);`

The `(axis>=3) ? 0 : 1.0` statement allows me to set the minumum value of the max travel to zero for axis A,B and C and with the change I made to the Limits.cpp-->limitsCheckTravel I can use rotary axes without soft limits 

Thank you very much for your help. It was very helpful !

nsiatras, 2021-03-19:
@bdring  and @MitchBradley  PS. Please consider this change for the project.
Using a rotary axis without soft limits is very useful for 4-Axis CNC milling  and makes the post processor work, for many cheap CAM software, easy as a walk in the park. 

Thanks again :)

bdring, 2021-03-19:
We will strongly consider it. I'll close this issue for now.

Here is an invite to our Discord server if you have future questions

https://discord.gg/Nzz9gWZV4K

closed 2021-03-19 by bdring

bdring, 2021-03-21:
I created a SoftLimit branch with the changes. I did not limit it to axes AB & C. 

Will make a PR