bdring/Grbl_Esp32 #832: User input buttons (hold, resume, reset)There is no change in the input of the high and low level of the pin.

txg5230, 2021-03-19:
The user input button does not respond

bdring, 2021-03-19:
We need more info. 
Tell us everything related to this.

Romain3481, 2021-03-24:
Hi !
I get the same issue, there is no effect when i push hold / start / rest button
Probe is connected with the same way and works well.
Thank for your help

bdring, 2021-03-24:
It works for me.

If you want more help with your problem, please answer the following questions.

- What build number of the firmware are you using?
- What hardware are you running on?
- What pins do you have defined for the inputs.

Also, paste your boot messages in the reply.

Romain3481, 2021-03-24:
Thanks for the fast reply
I downloaded the firmware yesterday 
I use a custom board , i put R/C filters on those inputs : probe, reset, start, hold.
Probe works very well, but not the other buttons (NO)
[image]
`#define PROBE_PIN               GPIO_NUM_16
#define COOLANT_MIST_PIN        GPIO_NUM_22

#define CONTROL_RESET_PIN       GPIO_NUM_4  // labeled Reset
#define CONTROL_FEED_HOLD_PIN   GPIO_NUM_2  // labeled Hold
#define CONTROL_CYCLE_START_PIN GPIO_NUM_0  // labeled Start`

I would like that the reset button fix alarm after limit switch reached, pause a job with the hold button, and resume it with start button.
Thank you !
17:03:02.501 -> [MSG:Connecting..]
17:03:02.968 -> [MSG:Connecting...]
17:03:03.473 -> [MSG:Connecting....]
17:03:03.513 -> [MSG:Connected with 192.168.1.122]
17:03:03.980 -> [MSG:Start mDNS with hostname:http://grblesp.local/]
17:03:04.026 -> [MSG:SSDP Started]
17:03:04.026 -> [MSG:HTTP Started]
17:03:04.026 -> [MSG:TELNET Started 23]
17:03:04.026 -> [MSG:Mist coolant on pin GPIO(22)]
17:03:04.026 -> [MSG:X  Axis limit switch on pin GPIO(36)]
17:03:04.026 -> [MSG:Y  Axis limit switch on pin GPIO(39)]
17:03:04.026 -> [MSG:Z  Axis limit switch on pin GPIO(34)]
17:03:04.026 -> [MSG:A  Axis limit switch on pin GPIO(35)]
17:03:04.026 -> [MSG:Probe on pin GPIO(16)]
17:03:04.026 -> 
17:03:04.026 -> Grbl 1.3a ['$' for help]

bdring, 2021-03-24:
You are missing the beginning of your boot messages

Romain3481, 2021-03-24:
8:20:07.276 -> [MSG:Grbl_ESP32 Ver 1.3a Date 20210311]
18:20:07.276 -> [MSG:Compiled with ESP32 SDK:v3.3.4-432-g7a85334d8]
18:20:07.276 -> [MSG:Using machine:RS-CNC32 Driver Board V2]
18:20:07.276 -> [MSG:Axis count 4]
18:20:07.276 -> [MSG:RMT Steps]
18:20:07.276 -> [MSG:Reset switch on pin GPIO(4)]
18:20:07.323 -> [MSG:Hold switch on pin GPIO(2)]
18:20:07.323 -> [MSG:Start switch on pin GPIO(0)]
18:20:07.323 -> [MSG:Init Motors]
18:20:07.323 -> [MSG:Reset via control pin]
18:20:07.323 -> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
18:20:07.323 -> Core 1 register dump:
18:20:07.323 -> PC      : 0x400d555a  PS      : 0x00060331  A0      : 0x800df52e  A1      : 0x3ffc08b0  
18:20:07.323 -> A2      : 0x00000000  A3      : 0xe0e952c4  A4      : 0x3f402384  A5      : 0x3ffc8d7c  
18:20:07.323 -> A6      : 0x3ffc0848  A7      : 0x00000016  A8      : 0x3ffc5e44  A9      : 0x00000010  
18:20:07.323 -> A10     : 0x00000000  A11     : 0xe0e952c4  A12     : 0x3ffc07ac  A13     : 0x3ffc8d7c  
18:20:07.370 -> A14     : 0x3ffc07ac  A15     : 0x0000001e  SAR     : 0x00000000  EXCCAUSE: 0x0000001c  
18:20:07.370 -> EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffff8  
18:20:07.370 -> Core 1 was running in ISR context:
18:20:07.370 -> EPC1    : 0x400d555a  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x4008ba34
18:20:07.370 -> 
18:20:07.370 -> ELF file SHA256: 0000000000000000
18:20:07.370 -> 
18:20:07.370 -> Backtrace: 0x400d555a:0x3ffc08b0 0x400df52b:0x3ffc08d0 0x40081574:0x3ffc08f0 0x40081671:0x3ffc0910 0x4008cdf1:0x3ffc0930 0x400d5698:0x3ffd1900 0x400d47a8:0x3ffd1930 0x400d2e47:0x3ffd1950 0x400fb2aa:0x3ffd1970 0x40091df6:0x3ffd1990
18:20:07.415 -> 
18:20:07.415 -> Rebooting...
18:20:07.977 -> 
18:20:07.977 -> [MSG:Grbl_ESP32 Ver 1.3a Date 20210311]
18:20:07.977 -> [MSG:Compiled with ESP32 SDK:v3.3.4-432-g7a85334d8]
18:20:07.977 -> [MSG:Using machine:RS-CNC32 Driver Board V2]
18:20:07.977 -> [MSG:Axis count 4]
18:20:07.977 -> [MSG:RMT Steps]
18:20:07.977 -> [MSG:Reset switch on pin GPIO(4)]
18:20:07.977 -> [MSG:Hold switch on pin GPIO(2)]
18:20:07.977 -> [MSG:Start switch on pin GPIO(0)]
18:20:07.977 -> [MSG:Init Motors]
18:20:07.977 -> [MSG:Global stepper disable pin:GPIO(15)]
18:20:07.977 -> [MSG:X  Axis Standard Stepper Step:GPIO(33) Dir:GPIO(25) Disable:None Limits(0.000,300.000)]
18:20:08.023 -> [MSG:Y  Axis Standard Stepper Step:GPIO(26) Dir:GPIO(27) Disable:None Limits(0.000,300.000)]
18:20:08.023 -> [MSG:Z  Axis Standard Stepper Step:GPIO(14) Dir:GPIO(12) Disable:None Limits(-300.000,0.000)]
18:20:08.023 -> [MSG:A  Axis Standard Stepper Step:GPIO(13) Dir:GPIO(32) Disable:None Limits(-300.000,0.000)]
18:20:08.023 -> [MSG:PWM spindle Output:GPIO(21), Enbl:GPIO(17), Dir:None, Freq:5000Hz, Res:13bits]
18:20:08.117 -> 
18:20:08.117 -> [MSG:Client Started]
18:20:08.165 -> [MSG:Connecting BOXROM]
18:20:08.165 -> [MSG:Connecting.]
18:20:08.633 -> [MSG:Connecting..]
18:20:08.914 -> [MSG:Connected with 192.168.1.122]
18:20:09.147 -> [MSG:Start mDNS with hostname:http://grblesp.local/]
18:20:09.147 -> [MSG:SSDP Started]
18:20:09.147 -> [MSG:HTTP Started]
18:20:09.147 -> [MSG:TELNET Started 23]
18:20:09.147 -> [MSG:Mist coolant on pin GPIO(22)]
18:20:09.193 -> [MSG:X  Axis limit switch on pin GPIO(36)]
18:20:09.193 -> [MSG:Y  Axis limit switch on pin GPIO(39)]
18:20:09.193 -> [MSG:Z  Axis limit switch on pin GPIO(34)]
18:20:09.193 -> [MSG:A  Axis limit switch on pin GPIO(35)]
18:20:09.193 -> [MSG:Probe on pin GPIO(16)]
18:20:09.193 -> 
18:20:09.193 -> Grbl 1.3a ['$' for help]

bdring, 2021-03-24:
It looks like you had a reset during boot.

18:20:07.323 -> [MSG:Reset via control pin]

This appears to have caused a reset. It might be due to multiple resets occurring very quickly. You might have some noise on that pin.

Also, GPIO_NUM_0 is not recommended for use. See the Wiki.

I would suggest testing with just one control input on GPIO_NUM_4. See it up for hold. Then check to see that it is being read using the ? command.

Romain3481, 2021-03-24:
I did a couple of most accurate tests :
Only the board is connected to a serial touchscreen on my desk.

With no modifications:
pin 0 - C. Start : works 1 second if i my finger stays pushed on the button and returns to HOLD:0 status
pin 2 - Hold -> nothing happen
pin 4 - Reset -> stop the job with the message "... stopped by reset pin" but don't remove the alarm when i trigger a limit switch

I removed all R/C filter ans ponted contacts:
pin 0 - C. Start : works 1 second if i my finger stays pushed on the button and returns to HOLD:0 status
pin 2 - Hold -> WORKS !!!!
pin 4 - Reset -> stop the job with the message "... stopped by reset pin" but don't remove the alarm when i trigger a limit switch

I soldered again the R/C filters but tested the functions one by one on pin 16 after flashing each time :
-C. Start : works 1 second if i my finger stays pushed on the button and returns to HOLD:0 status
- Hold -> WORKS !!!!
- Reset -> stop the job with the message "... stopped by reset pin" but don't remove the alarm when i trigger a limit switch

If you can help me again ^^
Thank you

bdring, 2021-03-24:
Are you using the WebUI for status? The response is 1 second in the best case. You should only have to push the button very quickly, then wait for the status.

Romain3481, 2021-03-25:
I'm not using the WebUI
Yes i push very quickly, when i told "Runs 1 second if i my finger stays pushed on the button and returns to HOLD:0 status", when i push quickly, it resumes just the time i stay pushed on it, if i push 2/10th of second, i see "run status" during this time, and once released, the status becomes HOLD:0 again and don't stays "RUN"

Romain3481, 2021-03-25:
Hi !
Any idea ?
I get crazy, i don't find the logic in my tests.
I understand that GPIO0 is not recommended, GPIO2 is better for output, but why get i the same issue on GPIO16 ?
Thx

bdring, 2021-03-25:
I don't know. I tested myself and it works.

Use the Platformio terminal to send ? very fast by holding down the key. Look at the Pn: values for input pin status while pushing the switches.

<Door:1|MPos:0.000,0.000,0.000|FS:0,0|Pn:DRHS|WCO:0.000,0.000,0.000>