bdring/Grbl_Esp32 #835: Ganged Y Axis not moving at equal steps/mm

ofek256, 2021-03-20:
Please only submit bugs for **latest** main or devt branches. You can check the version number in the startup messgaes and compare it to the version in [grbl.h](https://github.com/bdring/Grbl_Esp32/blob/main/Grbl_Esp32/src/Grbl.h)

Latest GRBL version on Fysetc E4 board with slightly modified config file to set A as Y2 motor.
Every time I try to move the Y axis, the two motor move at different steps/mm. When I switch the motors between the drivers the problem continues but switches motor, which is expected.

config for my board:

#define MACHINE_NAME            "FYSTEC E4 3D Printer Controller"

#define N_AXIS 3

#define TRINAMIC_RUN_MODE       TrinamicMode :: StealthChop
#define TRINAMIC_HOMING_MODE    TrinamicMode :: StealthChop

#define TMC_UART                UART_NUM_1
#define TMC_UART_RX             GPIO_NUM_21
#define TMC_UART_TX             GPIO_NUM_22   

#define X_TRINAMIC_DRIVER       2209
#define X_STEP_PIN              GPIO_NUM_27
#define X_DIRECTION_PIN         GPIO_NUM_26
#define X_RSENSE                TMC2209_RSENSE_DEFAULT
#define X_DRIVER_ADDRESS        1
#define DEFAULT_X_MICROSTEPS    16

#define Y_TRINAMIC_DRIVER       2209
#define Y_STEP_PIN              GPIO_NUM_33
#define Y_DIRECTION_PIN         GPIO_NUM_32
#define Y_RSENSE                TMC2209_RSENSE_DEFAULT
#define Y_DRIVER_ADDRESS        3
#define DEFAULT_Y_MICROSTEPS    16

#define Z_TRINAMIC_DRIVER       2209
#define Z_STEP_PIN              GPIO_NUM_14
#define Z_DIRECTION_PIN         GPIO_NUM_12
#define Z_RSENSE                TMC2209_RSENSE_DEFAULT
#define Z_DRIVER_ADDRESS        0
#define DEFAULT_Z_MICROSTEPS    16

#define Y2_TRINAMIC_DRIVER       2209
#define Y2_STEP_PIN              GPIO_NUM_16
#define Y2_DIRECTION_PIN         GPIO_NUM_17
#define Y2_RSENSE                TMC2209_RSENSE_DEFAULT
#define Y2_DRIVER_ADDRESS        3
#define DEFAULT_Y2_MICROSTEPS    16

#define X_LIMIT_PIN             GPIO_NUM_34
#define Y_LIMIT_PIN             GPIO_NUM_35
#define Z_LIMIT_PIN             GPIO_NUM_15
#define Y2_LIMIT_PIN             GPIO_NUM_36   // Labeled TB
#define PROBE_PIN               GPIO_NUM_39  // Labeled TE

// OK to comment out to use pin for other features
#define STEPPERS_DISABLE_PIN    GPIO_NUM_25

#define SPINDLE_TYPE            SpindleType::RELAY
#define SPINDLE_OUTPUT_PIN      GPIO_NUM_13  // labeled Fan
#define COOLANT_MIST_PIN        GPIO_NUM_2   // Labeled Hotbed
#define COOLANT_FLOOD_PIN       GPIO_NUM_4   // Labeled Heater

Any help appreciated, thank you!

bdring, 2021-03-20:
You have 2 motors at address #3

ofek256, 2021-03-20:
> You have 2 motors at address #3

Thank you, I changed it but both axis are still not synced.

bdring, 2021-03-20:
How much different are they? A little?...double?

What are your motor current settings?

ofek256, 2021-03-20:
After a bit of research, I found out that the Y2 driver's Rsense is supposed to be 0.22f, but when I change that in config and upload to the board I get the same result while running in SPI. In UART it seems to work but the X axis only moves clockwise for some reason while in UART so i'm trying to get it working in SPI:

#define MACHINE_NAME            "FYSETC E4 3D Printer Controller"

#define N_AXIS 3

#define TRINAMIC_RUN_MODE       TrinamicMode :: StealthChop
#define TRINAMIC_HOMING_MODE    TrinamicMode :: StealthChop

#define TMC_UART                UART_NUM_1
#define TMC_UART_RX             GPIO_NUM_21
#define TMC_UART_TX             GPIO_NUM_22   

#define X_TRINAMIC_DRIVER       2209
#define X_STEP_PIN              GPIO_NUM_27
#define X_DIRECTION_PIN         GPIO_NUM_26
#define X_RSENSE                0.11f
#define X_DRIVER_ADDRESS        1
#define DEFAULT_X_MICROSTEPS    16

#define Y_TRINAMIC_DRIVER       2209
#define Y_STEP_PIN              GPIO_NUM_33
#define Y_DIRECTION_PIN         GPIO_NUM_32
#define Y_RSENSE                0.11f
#define Y_DRIVER_ADDRESS        3
#define DEFAULT_Y_MICROSTEPS    16

#define Z_TRINAMIC_DRIVER       2209
#define Z_STEP_PIN              GPIO_NUM_14
#define Z_DIRECTION_PIN         GPIO_NUM_12
#define Z_RSENSE                0.11f
#define Z_DRIVER_ADDRESS        0
#define DEFAULT_Z_MICROSTEPS    16

#define Y2_TRINAMIC_DRIVER       2209
#define Y2_STEP_PIN              GPIO_NUM_16
#define Y2_DIRECTION_PIN         GPIO_NUM_17
#define Y2_RSENSE                0.22f
#define Y2_DRIVER_ADDRESS        2
#define DEFAULT_Y2_MICROSTEPS    16

#define X_LIMIT_PIN             GPIO_NUM_34
#define Y_LIMIT_PIN             GPIO_NUM_35
#define Z_LIMIT_PIN             GPIO_NUM_15
#define Y2_LIMIT_PIN            GPIO_NUM_36   // Labeled TB
#define PROBE_PIN               GPIO_NUM_39  // Labeled TE

// OK to comment out to use pin for other features
#define STEPPERS_DISABLE_PIN    GPIO_NUM_25

#define SPINDLE_TYPE            SpindleType::RELAY
#define SPINDLE_OUTPUT_PIN      GPIO_NUM_13  // labeled Fan
#define COOLANT_MIST_PIN        GPIO_NUM_2   // Labeled Hotbed
#define COOLANT_FLOOD_PIN       GPIO_NUM_4   // Labeled Heater

This is my current config.

> How much different are they? A little?...double?
> 
> What are your motor current settings?
Yeah, the Y2 motor does half the steps the Y1 motor does.
Microstepping is set to 16 in both, and current is 1A. 
Y is set to 400 steps per mm, and so is A if it changes anything.

bdring, 2021-03-20:
I have a Fytec board and it works fine with the machine definition included with Gbrl_ESP32

ofek256, 2021-03-20:
> I have a Fytec board and it works fine with the machine definition included with Gbrl_ESP32

According to this @ line 2238: https://github.com/FYSETC/FYSETC-E4/blob/main/examples/Ender-3/Configuration_adv.h
Rsense is supposed to be 0.22 for second motors. Either way, for some reason the change does nothing, and running UART makes the X motor only rotate one way. Maybe I have a faulty board?

ofek256, 2021-03-20:
Nevermind that, just managed to get it working! I really don't know what happened, but it now works while on UART. 
I'm attaching my config in case anyone ever runs into the same problem as me, but I highly doubt it.
Thanks for your help!

[fystec_e4.zip](https://github.com/bdring/Grbl_Esp32/files/6175707/fystec_e4.zip)

I'm closing this issue.

closed 2021-03-20 by ofek256

bdring, 2021-03-20:
Their documentation is incorrect. The controllers are built with 0.11 Ohm resistors. It is easy to check your board for that. The current sense resistors are located above the motor connectors.

The sense resistors are used for current regulation. The drivers are looking for a voltage drop across that resistor to sense current. If you specify 0.22 rather than 0.11 the current will be off by 2x. You might not be able to notice this in normal running.

The temperature and holding toque of the motor and driver will be off by 2x.

ofek256, 2021-03-20:
Well, it now works with 0.11f in Y2. I really don't know what caused the initial problem, then. I'll attach the updated file and remove the old one.
Thank you!

[fystec_e4.zip](https://github.com/bdring/Grbl_Esp32/files/6175867/fystec_e4.zip)