bdring/Grbl_Esp32 #837: Test drive Mode not working

atrejo, 2021-03-21:
By just compiling and loading to my esp32 doesn't work, software keeps restarting, I loaded many other sketches that work but grbl_esp32 doesn't work

[image]

keeps restarting, any idea on how to solve this fix?

bdring, 2021-03-21:
Are you sure the partition and other setting are all correct?

https://github.com/bdring/Grbl_Esp32/wiki/Compiling-with-Arduino-IDE

[image]

RobertoRmz, 2021-03-21:
Put a 10 uF capacitor on pin En so it can charge direct and follow the steps mentioned by bdring.

atrejo, 2021-03-30:
> Put a 10 uF capacitor on pin En so it can charge direct and follow the steps mentioned by bdring.

Sorry I didn't understand in which pin should I put the capacitor

RobertoRmz, 2021-03-30:
@atrejo  This is the image of how to connect a capacitor so that when loading the Firmware you are not pressing the button. 

[image]