bdring/Grbl_Esp32 #841: ESP32 reboot when asking a refresh of SD card list in WebUI

mstrens, 2021-03-25:
I just start testing the program. So it could be that I make a mistake.
I have a SD card attached to GRBL_ESP32. It contains some .nc files.
Using the serial interface, I can get the list of files with $SD/List command. Result is OK.
I can start the webUI and I can send move commands and see the status/position data reported by GRBL to "?" command.
Still, there are no files displayed in the tab foreseen to see the content of SD card.
If I press on the button refresh, I get an error popup (Connection error) and the Serial says that there is an error and the ESP32 reboot.
Here the `message:  
`ok
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400929b5  PS      : 0x00060033  A0      : 0x40094463  A1      : 0x3ffbd450  
A2      : 0x00050023  A3      : 0x00000001  A4      : 0x3ffca118  A5      : 0x3ffca118  
A6      : 0x00000001  A7      : 0x00000001  A8      : 0x00000001  A9      : 0x00000002  
A10     : 0x3ffbe314  A11     : 0x800d9070  A12     : 0x3ffca134  A13     : 0x3ffbceac  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x800d90b8  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x400929b5:0x3ffbd450 0x40094460:0x3ffbd470 0x40094416:0x00000000

Rebooting...

[MSG:Grbl_ESP32 Ver 1.3a Date 20210218]
[MSG:Compiled with ESP32 SDK:v3.3.4-441-gc43efe150]
[MSG:Using machine:Test Drive - Demo Only No I/O!]
[MSG:Axis count 3]
[MSG:Timed Steps]
[MSG:Init Motors]
[MSG:No spindle]

[MSG:Client Started]
[MSG:Connecting bbox2-58c4]
[MSG:Connecting.]
[MSG:Connecting..]
[MSG:Connecting...]
[MSG:Connecting....]
[MSG:Connecting.]
[MSG:Connecting..]
[MSG:Connected with 192.168.1.11]
[MSG:Connected with 192.168.1.11]
[MSG:Start mDNS with hostname:http://grblesp.local/]
[MSG:SSDP Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]

Grbl 1.3a ['$' for help]
`

mstrens, 2021-03-25:
Decoding the Backtrace gives:

Decoding stack results
0x400929b5: vTaskSwitchContext at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c line 2840

MitchBradley, 2021-03-25:
Try my code and see if it works.  That might give you a clue.

mstrens, 2021-03-25:
Note:  I am using version 1.0.5 of arduino ESP32 and I think I am even using the code from development on github

mstrens, 2021-03-25:
> Try my code and see if it works. That might give you a clue.

The issue is not in my firmware (GRBL controller) but in GRBL_ESP32 when using the webUI

MitchBradley, 2021-03-25:
I am currently rewriting the listing and running code so it works the same way for sd and spiffs.  It is possible that my rewrite will solve your problem - or maybe not.

mstrens, 2021-03-25:
Is your rewrite in the master branch from this site : https://github.com/MitchBradley/Grbl_Esp32   ?
In the mean time, I tried to reinstall version 1.0.5 of arduino ESP32 but GRBL_ESP32 did not compile. There are 2 conversion errors.
Then I reinstalled version 1.0.4 and then it compiled and I did not get anymore the connection error when I made a refresh of the SD card list. So the issue is related to the arduino ESP32 version.

MitchBradley, 2021-03-25:
No, my rewrite has not been pushed up anywhere yet.

Toaro, 2021-04-07:
I have the same problem, and I try also the branch https://github.com/bdring/Grbl_Esp32/tree/UARTFix but the problem is the same, I don't have access to the SD card. The error reported in serial monitor is:
`14:13:31.540 -> Grbl 1.3a ['$' for help]
14:14:05.950 -> ***ERROR*** A stack overflow in task clientCheckTask has been detected.
14:14:05.950 -> abort() was called at PC 0x40090010 on core 1
14:14:05.950 -> 
14:14:05.950 -> ELF file SHA256: 0000000000000000
14:14:05.950 -> 
14:14:05.950 -> Backtrace: 0x4008fd7c:0x3ffd1730 0x4008fff9:0x3ffd1750 0x40090010:0x3ffd1770 0x400928b7:0x3ffd1790 0x40094554:0x3ffd17b0 0x4009450a:0x00000000
14:14:05.950 -> 
14:14:05.950 -> Rebooting...
14:14:05.950 -> ets Jun  8 2016 00:22:57
14:14:05.950 -> 
14:14:05.950 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
14:14:05.950 -> configsip: 0, SPIWP:0xee
14:14:05.997 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
14:14:05.997 -> mode:DIO, clock div:1
14:14:05.997 -> load:0x3fff0018,len:4
14:14:05.997 -> load:0x3fff001c,len:1216
14:14:05.997 -> ho 0 tail 12 room 4
14:14:05.997 -> load:0x40078000,len:10944
14:14:05.997 -> load:0x40080400,len:6388
14:14:05.997 -> entry 0x400806b4
14:14:06.563 -> 
14:14:06.563 -> [MSG:Grbl_ESP32 Ver 1.3a Date 20210401]
14:14:06.563 -> [MSG:Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
14:14:06.563 -> [MSG:Using machine:PEN_LASER]
14:14:06.563 -> [MSG:Axis count 3]
14:14:06.563 -> [MSG:RMT Steps]
14:14:06.563 -> [MSG:Init Motors]
14:14:06.563 -> [MSG:Global stepper disable pin:GPIO(13)]
14:14:06.563 -> [MSG:X  Axis Standard Stepper Step:GPIO(12) Dir:GPIO(26) Disable:None Limits(-300.000,0.000)]
14:14:06.563 -> [MSG:Y  Axis Standard Stepper Step:GPIO(14) Dir:GPIO(25) Disable:None Limits(-300.000,0.000)]
14:14:06.611 -> [MSG:Z  Axis RC Servo Pin:27 Pulse Len(3276,6553) Limits(-100.000,0.000)]
14:14:06.611 -> [MSG:Servo Update Task Started]
14:14:06.611 -> [MSG:No spindle]
14:14:07.452 -> 
14:14:07.452 -> [MSG:Local access point GRBL_ESP started, 192.168.4.1]
14:14:07.452 -> [MSG:Captive Portal Started]
14:14:07.452 -> [MSG:HTTP Started]
14:14:07.499 -> [MSG:TELNET Started 23]
14:14:07.499 -> [MSG:X  Axis limit switch on pin GPIO(15)]
14:14:07.499 -> [MSG:Y  Axis limit switch on pin GPIO(4)]
14:14:07.499 -> 
14:14:07.499 -> Grbl 1.3a ['$' for help]
`

mstrens, 2021-04-07:
In my case, when I installed version 1.0.4 of arduino ESP32, the program GRBL_ESP32 works.

Toaro, 2021-04-07:
Yes, that is the solution... I downgrade to 1.0.5 and works. Thanks