bdring/Grbl_Esp32 #845: WiFi redirect error 500

YuraBz, 2021-03-29:
Hi, after recompilation grbl port the web console stopped working, tried to upload clear esp32 grbl image but the error remained.  
[image]

what could be the problem?  thanks

luc-github, 2021-03-29:
something is wrong in JSON generated by FW so the webUI do not understand it

MitchBradley, 2021-03-29:
We need to know which version of the firmware that you recompiled.

YuraBz, 2021-03-29:
the original version from here, at the first firmware upload everything worked, I fixed the step/for pin and after the web is no longer available, when changing back to the original version, too😕

MitchBradley, 2021-03-29:
Please be more specific than "original version".  I need to know the git branch name and the commit number.

YuraBz, 2021-03-29:
> Please be more specific than "original version". I need to know the git branch name and the commit number.

Sorry don’t know version or how get it, just download your zip file.

YuraBz, 2021-03-30:
17:05:33.024 -> ets Jun  8 2016 00:22:57
17:05:33.024 -> 
17:05:33.024 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
17:05:33.024 -> flash read err, 1000
17:05:33.024 -> ets_main.c 371 
17:05:33.401 -> ets Jun  8 2016 00:22:57
17:05:33.401 -> 
17:05:33.401 -> rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
17:05:33.401 -> configsip: 0, SPIWP:0xee
17:05:33.401 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
17:05:33.401 -> mode:DIO, clock div:1
17:05:33.401 -> load:0x3fff0018,len:4
17:05:33.401 -> load:0x3fff001c,len:1216
17:05:33.401 -> ho 0 tail 12 room 4
17:05:33.449 -> load:0x40078000,len:9720
17:05:33.449 -> ho 0 tail 12 room 4
17:05:33.449 -> load:0x40080400,len:6352
17:05:33.449 -> entry 0x400806b8
17:05:34.017 -> 
17:05:34.017 -> [MSG:Grbl_ESP32 Ver 1.3a Date 20210320]
17:05:34.017 -> [MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
17:05:34.017 -> [MSG:Using machine:ESP32_V4]
17:05:34.017 -> [MSG:Axis count 3]
17:05:34.017 -> [MSG:RMT Steps]
17:05:34.017 -> [MSG:Hold switch on pin GPIO(36)]
17:05:34.017 -> [MSG:Init Motors]
17:05:34.017 -> [MSG:Global stepper disable pin:GPIO(13)]
17:05:34.017 -> [MSG:X  Axis Standard Stepper Step:GPIO(12) Dir:GPIO(26) Disable:None Limits(0.000,300.000)]
17:05:34.017 -> [MSG:Y  Axis Standard Stepper Step:GPIO(14) Dir:GPIO(25) Disable:None Limits(0.000,300.000)]
17:05:34.063 -> [MSG:Z  Axis Standard Stepper Step:GPIO(27) Dir:GPIO(33) Disable:None Limits(-300.000,0.000)]
17:05:34.063 -> [MSG:PWM spindle Output:GPIO(17), Enbl:GPIO(16), Dir:None, Freq:5000Hz, Res:13bits]
17:05:34.862 -> 
17:05:34.862 -> [MSG:Local access point GRBL_ESP started, 192.168.0.1]
17:05:34.862 -> [MSG:Captive Portal Started]
17:05:34.909 -> [MSG:HTTP Started]
17:05:34.909 -> [MSG:TELNET Started 23]
17:05:34.909 -> [MSG:Flood coolant on pin GPIO(18)]
17:05:34.909 -> [MSG:Mist coolant on pin GPIO(19)]
17:05:34.909 -> [MSG:X  Axis limit switch on pin GPIO(21)]
17:05:34.909 -> [MSG:Y  Axis limit switch on pin GPIO(22)]
17:05:34.909 -> [MSG:Z  Axis limit switch on pin GPIO(23)]
17:05:34.909 -> [MSG:Probe on pin GPIO(32)]
17:05:34.909 -> 
17:05:34.909 -> Grbl 1.3a ['$' for help]

MitchBradley, 2021-03-30:
I cannot reproduce this error.  I did the following:

1. Downloaded and extracted the zip file from https://github.com/bdring/Grbl_Esp32/archive/refs/heads/main.zip
2. In VSCode, opened the folder where I extracted that zip
3. Used PlatformIO in VSCode to compile and upload, without changing anything in the source tree.  That resulted in a "test drive" firmware image.
4. Browsed to the Station mode URL listed in the startup messages (it was in station mode because I had previously set that configuration in the saved preferences).  Everything worked correctly.  I was able to list files on SD and SPIFFS, delete and re-upload the index.html.gz file, etc.
5. Seeing that you are in AP mode, I switched to AP mode and tried again, connecting my PC to the GRBL_ESP AP and browsing to the 192.168.0.1 AP mode address.  Again, everything worked.
6. Seeing the "Using machine:ESP32_V4" line in your startup messages, I recompiled with the machine definition "Machines/3axis_v4.h" and uploaded.
7. Again, everything worked correctly, including all of the SPIFFS filesystem operations listed above.
8. I see that your machine configuration is different from the stock 3axis_v4.h .  The pins are different.  Here are the startup messages from my compilation of the stock 3axis_v4.h

ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8

[MSG:Grbl_ESP32 Ver 1.3a Date 20210320]
[MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
[MSG:Using machine:ESP32_V4]
[MSG:Axis count 3]
[MSG:RMT Steps]
[MSG:Init Motors]
[MSG:Global stepper disable pin:GPIO(13)]
[MSG:X  Axis Standard Stepper Step:GPIO(12) Dir:GPIO(14) Disable:None Limits(20.000,320.000)]
[MSG:Y  Axis Standard Stepper Step:GPIO(26) Dir:GPIO(15) Disable:None Limits(0.000,310.000)]
[MSG:Z  Axis Standard Stepper Step:GPIO(27) Dir:GPIO(33) Disable:None Limits(-300.000,0.000)]
[MSG:Warning: LASER_OUTPUT_PIN not defined]

[MSG:Local access point GRBL_ESP started, 192.168.0.1]
[MSG:Captive Portal Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]
[MSG:Flood coolant on pin GPIO(25)]
[MSG:Mist coolant on pin GPIO(21)]
[MSG:X  Axis limit switch on pin GPIO(17)]
[MSG:Y  Axis limit switch on pin GPIO(4)]
[MSG:Z  Axis limit switch on pin GPIO(16)]
[MSG:Probe on pin GPIO(32)]

Grbl 1.3a ['$' for help]

MitchBradley, 2021-03-30:
Please test with a standard unmodified configuration, preferably test_drive.h

Also tell us your compilation method.  If you are using Arduino IDE, there have been recent reports of problems with the 1.0.5 and 1.0.6 versions of the Arduino ESP32 framework.  People have reported that reverting to 1.0.4 made the problems go away.

YuraBz, 2021-03-30:
> Please test with a standard unmodified configuration, preferably test_drive.h
>
> Also tell us your compilation method. If you are using Arduino IDE, there have been recent reports of problems with the [...]

thanks.  I downloaded firmwareto another esp32, it works. 
 i will try to restore esp to factory