bdring/Grbl_Esp32 #848: Expected command letter and Bad Number format Errors Happen randomly.

nsiatras, 2021-03-29:
**What version of the firmware are you using?**
The latest

**Is the problem repeatable?**
Yes

**Under what conditions does the bug occur?**

I have the NodeMCU on my desk, there are no connections to stepper drivers or anything else. It is just connected to the USB and I am using serial connection to cycle the gcode.

I noticed that when the gcode runs fast I get back the Errors **1** and **2**
1 --> Expected command letter: G-code words consist of a letter and a value. Letter was not found.
2 --> Bad number format: Missing the expected G-code word value or numeric value format is not valid.

When I say that the Gcode runs fast I mean that there are lines that are executed quickly (small movements).
This also happens when I set the board to check mode by sending the $C command.

The errors are never in the same line. If I cycle the gcode now I might get the Error 1 on Line 2452 and if I cycle it again I get Error 1 in line 19502. So.. I get error but never in the same line...

Any ideas why is this happening ?

Here is my Setup
```
$0 = 10    (Step pulse time, microseconds)
$1 = 255    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 4    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 1    (Status report options, mask)
$11 = 0.025    (Junction deviation, millimeters)
$12 = 0.005    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 0    (Hard limits enable, boolean)
$22 = 0    (Homing cycle enable, boolean)
$23 = 3    (Homing direction invert, mask)
$24 = 500.000    (Homing locate feed rate, mm/min)
$25 = 1500.000    (Homing search seek rate, mm/min)
$26 = 250.000    (Homing switch debounce delay, milliseconds)
$27 = 2.000    (Homing switch pull-off distance, millimeters)
$30 = 1000.000    (Maximum spindle speed, RPM)
$31 = 0.000    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 400.000    (X-axis travel resolution, step/mm)
$101 = 400.000    (Y-axis travel resolution, step/mm)
$102 = 400.000    (Z-axis travel resolution, step/mm)
$103 = 26.666   
$104 = 100.000   
$105 = 100.000   
$110 = 4000.000    (X-axis maximum rate, mm/min)
$111 = 4000.000    (Y-axis maximum rate, mm/min)
$112 = 2500.000    (Z-axis maximum rate, mm/min)
$113 = 14400.000   
$114 = 1000.000   
$115 = 1000.000   
$120 = 80.000    (X-axis acceleration, mm/sec^2)
$121 = 80.000    (Y-axis acceleration, mm/sec^2)
$122 = 60.000    (Z-axis acceleration, mm/sec^2)
$123 = 720.000   
$124 = 200.000   
$125 = 200.000   
$130 = 302.000    (X-axis maximum travel, millimeters)
$131 = 202.000    (Y-axis maximum travel, millimeters)
$132 = 180.000    (Z-axis maximum travel, millimeters)
$133 = 0.000   
$134 = 300.000   
$135 = 300.000   
```

truglodite, 2021-03-30:
I have also been troubleshooting this problem. It fails error 1,2, etc indicating malformed gcode. This is even with test_drive.h, bare esp32d devkitC, and any gcode I throw at it.

```
ets Jun  8 2016 00:22:57
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
**** Connected to COM5 @ 115200 baud ****
[MSG:Grbl_ESP32 Ver 1.3a Date 20210320]
[MSG:Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:Using machine:Test Drive - Demo Only No I/O!]
[MSG:Axis count 3]
[MSG:Timed Steps]
[MSG:Init Motors]
[MSG:No spindle]
[MSG:Local access point GRBL_ESP started, 192.168.4.1]
[MSG:Captive Portal Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]
Grbl 1.3a ['$' for help]
>>> $$
$0 = 3    (Step pulse time, microseconds)
$1 = 250    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 0    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 1    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 1    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 0    (Hard limits enable, boolean)
$22 = 0    (Homing cycle enable, boolean)
$23 = 3    (Homing direction invert, mask)
$24 = 200.000    (Homing locate feed rate, mm/min)
$25 = 2000.000    (Homing search seek rate, mm/min)
$26 = 250.000    (Homing switch debounce delay, milliseconds)
$27 = 1.000    (Homing switch pull-off distance, millimeters)
$30 = 1000.000    (Maximum spindle speed, RPM)
$31 = 0.000    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 100.000    (X-axis travel resolution, step/mm)
$101 = 100.000    (Y-axis travel resolution, step/mm)
$102 = 100.000    (Z-axis travel resolution, step/mm)
$103 = 100.000   
$104 = 100.000   
$105 = 100.000   
$110 = 1000.000    (X-axis maximum rate, mm/min)
$111 = 1000.000    (Y-axis maximum rate, mm/min)
$112 = 1000.000    (Z-axis maximum rate, mm/min)
$113 = 1000.000   
$114 = 1000.000   
$115 = 1000.000   
$120 = 200.000    (X-axis acceleration, mm/sec^2)
$121 = 200.000    (Y-axis acceleration, mm/sec^2)
$122 = 200.000    (Z-axis acceleration, mm/sec^2)
$123 = 200.000   
$124 = 200.000   
$125 = 200.000   
$130 = 300.000    (X-axis maximum travel, millimeters)
$131 = 300.000    (Y-axis maximum travel, millimeters)
$132 = 300.000    (Z-axis maximum travel, millimeters)
$133 = 300.000   
$134 = 300.000   
$135 = 300.000   
ok
```

To troubleshoot this, I tried on 6 different esp32 boards, from 3 different vendors purchased 2 different times. I tried bCNC and UGS. I scoped TX and RX directly on the esp. The min pulses at 115k were 8.68usec on the money, and the RS232 decoder saw all the data fine. No spikes on the 5V line either. At first I thought maybe my DIY expansion modules were burning esp boards, but the last 3 have never been plugged into anything but my PC usb port for testing this.

I guess it might be some recent update to one of the dependencies that broke something. I'm using the absolute latest to compile (per the wiki). That also includes that latest Arduino release, latest nightly (as of a few hours ago), and a fully updated PIO setup. I wish I knew where to start finding a bug.

nsiatras, 2021-03-30:
@truglodite I also thought that my board was damaged so I purchased other 2 boards.
The bug exists on all 3 of them..

Have you noticed any unusual machine moves while cycling gcode? Because I did...

truglodite, 2021-03-30:
@nsiatras Yes in fact half the time when testing the router gcode it would move off along one axis for a very long time (crashing if plugged in without soft stops or hard stops). 

Last week I fired up my 6-pack the first time and had that runaway along the Y twice in a row trying to mill some pcb's. Then I dropped in a new esp module, and finished several cuts no problem. Then I popped in the laser, and realized I had to recompile for laser mode M4. After that I have dealt with this issue. There was an update to the esp32 board files 4 days ago. Maybe it lies in there.

nsiatras, 2021-03-30:
That is exactly what happened to me 3 times already. It moves one axis (mostly X) to the max and crashes.
I have the latest version but I do not know if the bug existed in the previous one. I didn't noticed

truglodite, 2021-03-30:
'Duino board versions 1.0.5 and 1.0.6 are fairly new. I tried 1.0.4 and sure enough it is working.

nsiatras, 2021-03-30:
I just downgraded to to v1.0.4 and I managed to cycle a gcode without any errors
Thank you @truglodite !

[image]

truglodite, 2021-03-30:
Now, back to cutting stuff... ;)

On a side note, I am not sure what changes to PIO go back to 1.0.4. I tried `platform = espressif32@>=1.0.4,<1.0.5` in platform.ini but that fails.

bdring, 2021-03-30:
Was the WebUI also open when you got errors?

nsiatras, 2021-03-30:
No. I have bluetooth and Wifi disabled.
I used the machine more than 8 hours today. It still runs a 4 axis gcode.
Still no errors

truglodite, 2021-03-30:
@bdring The errors with 1.0.6 were happening with or without wifi enabled, and with or without me poking around in the webui. With 1.0.4 I can play in the webui during a demanding etch and have no issues.

bdring, 2021-03-30:
What serial sender were you using and can you upload an example file that fails.
Also please confirm that an unaltered test_drive.h machine definition fails?

nsiatras, 2021-03-30:
I use SourceRabbit GCode Sender. I also tested with Universal GCode Sender and I got errors too.

Since I downgraded the ESP32 library to version 1.0.4 the errors disappear and my machine works 100% correct.

I doen't matter of the GCode. I got errors with any gcode made by ESTLCam, SprutCAM, JSCut, DeskProto etc. but I am attaching a gcode for you to see
[Roughing.zip](https://github.com/bdring/Grbl_Esp32/files/6231978/Roughing.zip)

nsiatras, 2021-03-30:
PS. I never used the test_drive.h machine

bdring, 2021-03-30:
@nsiatras Please post your machine definition file. That might help me find the problem.

truglodite, 2021-03-31:
Same here, no changes to the code using test_drive.h. Any old gcode will do it; well something more than a square maybe, but typically it fails anywhere from immediately to around 3min max. Hence we could not complete any gcodes while troubleshooting this.

bdring, 2021-04-02:
This issue has been confirmed to be a bug in recent esp32 library code. 1.0.4 works, but not after that. An issue was created at the Espressif github.

https://github.com/espressif/arduino-esp32/issues/5005

truglodite, 2021-04-02:
Thanks Bart... and Mitch for figuring this one out. I subbed the esp32 issue and am looking forward to testing the fix out. For now I have added `platform = espressif32@3.0.0 ;workaround for 1.0.5+ breaking serial` to platform.ini release environment.

Avataar120, 2021-04-21:
Thanks a lot, you saved my life :)
Going back to library code v1.0.4 is working fine ! 
Just had to modify WebSocketsClient.cpp as setInsecure() is not existing.

Loking forward for the fix in ESP32 > 1.0.4 !

MitchBradley, 2021-04-22:
The Arduino fix is no longer necessary. We changed Grbl_Esp32 to use a UART driver from the ESP-IDF SDK instead of the Arduino framework driver. With that change, Grbl_Esp32 works with any version of the Arduino framework.

Avataar120, 2021-04-22:
> The Arduino fix is no longer necessary. We changed Grbl_Esp32 to use a UART driver from the ESP-IDF SDK instead of the Arduino framework driver. With that change, Grbl_Esp32 works with any version of the Arduino framework.

Thanks a lot, I can confirm it works even at very high baud rate !
You rock !

closed 2021-04-22 by MitchBradley