bdring/Grbl_Esp32 #851: step servo support for Z axis

prak1111, 2021-04-02:
**Please describe the feature you would like implemented**
Please add support for step servo for z axis.
**Why do you think this would improve Grbl_ESP32?**
This will help to remain z axis in position in event of power failure as step servo have builtin brake to prevent position loss
**What do you need the feature for?**
This feature is most useful for those who face a lot of power outage issues as moving spindle make the unwanted hole on the running cnc project.
**Will this feature appear to a lot of users?**
this feature really matters where power outage(many times) is common like developing countries.

bdring, 2021-04-02:
What is a step servo. Provide links and documentation please.

prak1111, 2021-04-14:
Hello,
Sorry for delay. Here is the link for hybrid servo and diver- [https://robokits.co.in/automation-control-cnc/leadshine-motor-drive/hybrid-servo-motor-2-phase-70kg-cm-w-t-es-d1008-leadshine-drive](url)
Link for driver user manual [https://robokits.download/downloads/Leadshine_ES-D808.pdf](url)

bdring, 2021-04-14:
Any driver, like that, that accepts step and direction inputs should work.

prak1111, 2021-04-14:
I thought other input/output ports needs to be controlled by the controller.