bdring/Grbl_Esp32 #853: Manual mode, can xyz only move, can it move continuously

txg5230, 2021-04-02:


bdring, 2021-04-02:
Please describe what you cannot do in detail please.

CNC machine axes are designed to move from point to point. They cannot move continuously.

txg5230, 2021-04-02:
My machine tool stroke x1300mm, y2500mm, z120mm. point-to-point movement is very inconvenient, can you add continuous movement function in manual mode

bdring, 2021-04-03:
Do you mean continuous jogging? That can be done in the gcode sender. Some do it well like Grbl Plotter.  The WebUI is not fast enough to do it effectively.

MAK-1966, 2022-02-21:
Hi, I need the same continuous jogging function
can you add a dedicated pin for jog like home or limit pin
for better , faster , preciser jog function 
thank you

MitchBradley, 2022-02-21:
Grbl_Esp32 has been superseded by FluidNC.  We are not longer doing any development on Grbl_Esp32.

closed 2022-02-21 by MitchBradley