bdring/Grbl_Esp32 #858: FPGA cnc control

hstarmans, 2021-04-08:
Hey all,

I am working on controlling motors with a FPGA.  You can see my working setup [here](https://youtu.be/-0uB2MydtrE) . 
The video is not the greatest but it shows where I am at. Code can be found here [FPGAG](https://github.com/hstarmans/FPGAG).
The code is written in python and gets converted to verilog/vhdl using nmigen. Yosys/Nextpnr translate the code to the ICE FPGA chip. FPGAs offer a lot of advantages.  The motors are sampled at 1 MHz and the maximum pulse frequency is 500 KHz.
This can, however be changed.  The max I can reach now is probably 25 MHz. 
I am mainly interested in FPGAs as I am building a laser scanner.   

Best,

Rik

MitchBradley, 2021-04-08:
You probably already know this but Raspberry Pi 4 has an FPGA that drives the GPIOs.  That might make a nice deployment platform.

hstarmans, 2021-04-09:
Mitch, I am not aware of a FPGA already present on the Raspberry pi 4. I know a Zynqberry exists, which is similar to a Raspberry which does have a FPGA on the board.

MitchBradley, 2021-04-09:
It seems like that announcement about an FPGA on the Pi 4 was an April's Fools joke.  https://www.mickmake.com/post/the-raspberry-pi-4-has-landed-a-sneak-peak-prototype-review/ Looks like I got fooled.  The [April Fools] tag was omitted in the early posts.

atlaste, 2021-04-10:
The more serious MESA cards for LinuxCNC do have FPGA support.

chenleigang, 2022-02-05:
This is a very good project. Although I can't understand it, I still like it very much

hstarmans, 2022-02-05:
thank @chenleigang, I am actually replacing my motors with [PCB motors](https://kevinlynagh.com/pcb-stepper/). This will make the tech a lot cheaper and more accessible.  I am working on replacing the raspberry pi with a ESP32. So this would also make it easier to integrate it with even this esp32 project.