bdring/Grbl_Esp32 #860: Splitting 74AHC595 8 bit registers into 2x 4 bit registers using 2x 74AHC194

edcasati, 2021-04-10:
I am working on a 6 axis robot arm build, and would like to use the ESP32 GRBL system as the main controller.
The common design runs the stepper wires and limit sensors through the arm base, and you end up with over 30 wires coming out the bottom of the arm. Big mess of wires.
I would like to be able to just run a power cable up the arm, plus a control wire for all the stepper motors. I would move the stepper drivers to next to each motor.
Bdring's SPI 6 axis control board manages the 6 steppers via 74AHC595 8 bit shift registers, daisy chained via I2S, each one feeding two steppers.
Does anyone familiar with that setup see any problem with using twice as many 4 bit shift registers, one per stepper, instead of the 8 bit shift registers feeding two steppers each? I can't see why not... but I wonder about the signal timing. No register addressing takes place (they all become active simultaneously on a common clock pulse) so I can't see why it wouldn't work.
Any thoughts on my idea?

MitchBradley, 2021-04-10:
With '194, as you shift in the bits, the bits from other channels will appear on the outputs.  The output state will be correct only briefly.  With '595, the value is transferred to an output latch during the brief period when the shift register outputs contain the correct channel values.  If you use '194, you would need external registers to duplicate that shift-then-transfer behavior.

edcasati, 2021-04-10:
Hmmm... I suspected there would be a catch. Thanks for the quick reply.
I wonder if there is a 4 bit equivalent in function to the 74AHC595. Time to explore digikey some more.
Padding would be another option, but that would mean branching the code, something I will not do.

MitchBradley, 2021-04-10:
If you find one, let me know.  I did a lot of searching during the initial design and did not find anything.  You could use '595 and waste some outputs.

That said, if you are thinking about running the I2S signals off-board, you are in for a world of problems.  This is deep EE stuff; timing, signal integrity, grounding, etc.  I2S is just not intended for long-haul use.  To go off-board, you need an interconnection method that is designed for noise immunity.  Typically, for high speed, that is some form of differential signaling like the techniques used in Ethernet, USB, and HDMI.  For low speed, optocouplers are good.

edcasati, 2021-04-10:
OK. It was a thought, but I asked because I obviously had some doubts. Time to rethink my 'design'.

Another option that I'm considering is having multiple processors, one for each segment, sub processing the GRBL commands. Each one would be fed an instruction to generate X number of pulses at a specific rate. The kinematics of acceleration and deceleration are a complication.

The general concept came to me from this video:
(30) Expanding TONS of Switches/LEDs with Arduino (Teensy + ATtiny) and Controlling over MIDI! - YouTube<https://www.youtube.com/watch?v=RpF947y642c&t=553s&ab_channel=KevinDarrah>

It would seem a good way to get two-way communication between nodes, allowing data such as encoder counts, limit switch interrupts, excess load signals such as that that a TMC2209 can provide, etc.  With the controller centralized, the wiring becomes unmanageable otherwise.

But it seems that a code rewrite from ground up might be necessary.

MitchBradley, 2021-04-10:
MIDI would probably work.  DMX is another alternative.

closed 2021-04-10 by MitchBradley