bdring/Grbl_Esp32 #861: Tiny VFD bug

atlaste, 2021-04-10:
I'm just writing this down so it doesn't get lost. Haven't traced the bug back yet.

While testing on my H2A (P2 Series Inverter) VFD, this is the behavior:

- start job
- finish job at -say- 24k rpm
- do some stuff, don't use spindle
- start new job at 24k rpm

At this point, it appears that the spindle sync doesn't kick in, and the job starts running immediately. The issue is probably that M5 should set `sync_rpm = 0` somewhere, but doesn't do that.

bdring, 2021-04-10:
Huanyang...

Started job at 8k
Finished job
Did some stuff
Started same job...worked fine

atlaste, 2021-04-11:
Thanks for testing. I think the issue is in the H2A code, probably some quirk in the protocol I don't understand. Haven't figured it out yet...

atlaste, 2021-04-11:
VFD_DEBUG_MODE output:

S9172 < ok
M3 S9172 < [MSG:Syncing RPM. Requested 9172, current 0]
[MSG:Syncing RPM. Requested 9172, current 0]
[MSG:Syncing RPM. Requested 9172, current 1195]
[MSG:Syncing RPM. Requested 9172, current 2479]
[MSG:Syncing RPM. Requested 9172, current 3771]
[MSG:Syncing RPM. Requested 9172, current 5046]
[MSG:Syncing RPM. Requested 9172, current 5691]
[MSG:Syncing RPM. Requested 9172, current 6967]
[MSG:Syncing RPM. Requested 9172, current 8258]
ok
M5 < ok
M3 S9172 < ok
M5 < ok
M3 S9172 < ok
M5 < ok

atlaste, 2021-04-14:
Implemented in atlaste/H2ASpindleSyncFix . Once tested I'll make a PR.

marcow1601, 2021-04-16:
So I ran into the exact same issue as described by @atlaste but with my Yalang YL620 type VFD. Ran a job at 24k RPM, sync worked perfectly, started another job at 24k, but the machine didn't wait for the spindle to get up to speed anymore.

I have now implemented the fix proposed in atlaste/H2ASpindleSyncFix. Tested with 4 subsequent jobs at 24k RPM and this seems to have fixed the issue for me.

I will open a pull-request for my YL620 VFD driver.

atlaste, 2021-04-16:
@marcow1601 I'll update both implementations in the PR, or just add it to the VFD code instead. No need to make another PR.

bdring, 2021-04-19:
PR #868 has been merged

closed 2021-04-19 by bdring