bdring/Grbl_Esp32 #862: Candle Heightmap Probing Not Working

Teselation, 2021-04-10:
### What version of the firmware are you using?
Grbl_Esp32: 6c8c613? (no idea i downloaded the latest thing from GitHub)
I tried Candle: 1.1.7  1.2.12b  1.2.13b
Arduino grbl: 1.1h

### Is the problem repeatable?
Yes it never works.

### Weird inconvenience:
Grbl_Esp32 has to have $6 set to 1 to probe (work) like Arduino grbl with $6=0.

### Under what conditions does the bug occur?
Probing button (inside Candle) works correctly.
When i tried to probe a heightmap machine moved as usual (exactly like it would with Arduino) but the results on each probe **(heightmap results) were always zero** independently from the height at which the probe detected the surface.
I don't know whether this is a problem with Candle or Grbl_Esp but it worked fine with Arduino.

**Recap:** Probing "Heightmap" does not work.

### Additional info:
Here i found a random video on youtube where you can see the box with the numbers.
For me all of them are 0.00 independent from where the probe stops.
https://youtu.be/7TRqrLqnKa0?t=255

Probe button that works fine is second one on the right (magnifying glass with arrow pointing down)
https://youtu.be/7TRqrLqnKa0?t=59

Again probing heightmap feature worked fine with Arduino.

Im not sure about that but it seems like Esp32 returns some additional information (in comparison to Arduino) after probing.

probe pin D21
all XYZABC axis and limit switches enabled

#### I need a quick fix to get my machine milling again after i rewired it for esp32.

```C++
```

bdring, 2021-04-11:
I did a quick test. It worked fine for me.

The arcs_arrow.nc file in the tests folder was used.

I did not have a pcb to use, so I just used a manual switch to simulate the touch.

Can you paste the console test from a filed run?

[image]

Teselation, 2021-04-11:
I hope that you wanted text from console because I don't know what console test is.
Im using USB for communication.
And there isn't a lot going one in the console while Heightmap probing.

### This is Heightmap 1.1.7
[image]

### This is probe (using Z-probe button)
[image]

### This is Heightmap run inside newest beta version 1.2.13b
[image]

Then i tried testing it on another computer and i accidentally connected pin D12  Xdir as a probe instead of D21 (i was looking upside down at the board) luckily nothing was damaged.

So i did my test nothing had changed (empty console (on another computer)).
While scrubbing through the settings i found:
**Show G-code program commands** checkbox
So here are the results after i made sure that this was checked:
[image]
[CTRL+X] < Grbl 1.3a ['$' for help]
S9600 < ok
G21G90F10G0Z1 < ok
G0X0Y0 < ok
G38.2Z-1 < [MSG:Found]
[PRB:0.000,0.000,-0.064,0.000,0.000,0.000:1]
ok
G0Z1 < ok
G0X0.000Y0.000 < ok
G38.2Z-1 < [MSG:Found]
[PRB:0.000,0.000,1.000,0.000,0.000,0.000:1]
ok
G0Z1 < ok
G0X50.000Y0.000 < ok
G38.2Z-1 < [MSG:Found]
[PRB:50.000,0.000,-0.139,0.000,0.000,0.000:1]
ok
G0Z1 < ok
G0X100.000Y0.000 < ok
G38.2Z-1 < [MSG:Found]
[PRB:100.000,0.000,-0.331,0.000,0.000,0.000:1]
ok
G0Z1 < ok
G0X100.000Y39.495 < ok
G38.2Z-1 < [MSG:Found]
[PRB:100.000,39.495,-0.207,0.000,0.000,0.000:1]
ok
G0Z1 < ok
G0X50.000Y39.495 < ok
G38.2Z-1 < [MSG:Found]
[PRB:50.000,39.495,-0.145,0.000,0.000,0.000:1]
ok
G0Z1 < ok
G0X0.000Y39.495 < ok
G38.2Z-1 < [MSG:Found]
[PRB:0.000,39.495,-0.181,0.000,0.000,0.000:1]
ok
G0Z1 < ok
G0X0.000Y78.990 < ok
G38.2Z-1 < [MSG:Found]
[PRB:0.000,78.990,-0.201,0.000,0.000,0.000:1]
ok
G0Z1 < ok
G0X50.000Y78.990 < ok
G38.2Z-1 < [MSG:Found]
[PRB:50.000,78.990,-0.083,0.000,0.000,0.000:1]
ok
G0Z1 < ok
G0X100.000Y78.990 < ok
G38.2Z-1 < [MSG:Found]
[PRB:100.000,78.990,-0.034,0.000,0.000,0.000:1]
ok
G0Z1 < ok

### This is Arduino
[image]
[CTRL+X] < Grbl 1.1h ['$' for help]
G92Z0 < ok
$# < [G54:-102.000,-53.000,-90.200]
[G55:0.000,0.000,0.000]
[G56:0.000,0.000,0.000]
[G57:0.000,0.000,0.000]
[G58:0.000,0.000,0.000]
[G59:0.000,0.000,0.000]
[G28:0.000,0.000,0.000]
[G30:0.000,0.000,0.000]
[G92:0.000,0.000,90.200]
[TLO:0.000]
[PRB:0.000,0.000,0.000:0]
ok
G92X0Y0 < ok
$# < [G54:-102.000,-53.000,-90.200]
[G55:0.000,0.000,0.000]
[G56:0.000,0.000,0.000]
[G57:0.000,0.000,0.000]
[G58:0.000,0.000,0.000]
[G59:0.000,0.000,0.000]
[G28:0.000,0.000,0.000]
[G30:0.000,0.000,0.000]
[G92:102.000,53.000,90.200]
[TLO:0.000]
[PRB:0.000,0.000,0.000:0]
ok
G21G90F10G0Z1 < ok
G0X0Y0 < ok
G38.2Z-1 < [PRB:0.000,0.000,0.385:1]
ok
G0Z1 < ok
G0X0.000Y0.000 < ok
G38.2Z-1 < [PRB:0.000,0.000,0.623:1]
ok
G0Z1 < ok
G0X50.000Y0.000 < ok
G38.2Z-1 < [PRB:49.998,0.000,0.412:1]
ok
G0Z1 < ok
G0X100.000Y0.000 < ok
G38.2Z-1 < [PRB:100.000,0.000,-0.283:1]
ok
G0Z1 < ok
G0X100.000Y39.495 < ok
G38.2Z-1 < [PRB:100.000,39.495,0.030:1]
ok
G0Z1 < ok
G0X50.000Y39.495 < ok
G38.2Z-1 < [PRB:49.998,39.495,0.431:1]
ok
G0Z1 < ok
G0X0.000Y39.495 < ok
G38.2Z-1 < [PRB:0.000,39.495,0.370:1]
ok
G0Z1 < ok
G0X0.000Y78.990 < ok
G38.2Z-1 < [PRB:0.000,78.989,0.223:1]
ok
G0Z1 < ok
G0X50.000Y78.990 < ok
G38.2Z-1 < [PRB:49.998,78.989,0.586:1]
ok
G0Z1 < ok
G0X100.000Y78.990 < ok
G38.2Z-1 < [PRB:100.000,78.989,0.597:1]
ok
G0Z1 < ok

For my unqualified eye it looks like all of 6 axes are messing with the Candle.
Although the probe button itself works.
Might be some bug inside Candle's code getting firs value from the right while probing Heightmap

bdring, 2021-04-11:
I agree. it appears to run the macro fine, but is having trouble with the 6 axis response. I suggest contacting Candle or changing your machine definition to 3 axis.

Teselation, 2021-04-11:
Thanks for answering so quickly. I will talk to Candle about that Heightmap problem but there is still this weird $6 reverse probe going on (which i mentioned earlier as a "Weird inconvenience:") and I don't know if this is supposed to work like that (throwing errors when $6 = 0).
When i moved from Arduino i wasted some time troubleshooting my probe (checking wires, schematic, gpio pin, measuring voltages) and it turned out that I needed to set $6 to 1. It works now but there might be other people having trouble with it.
Or maybe it's just my board?

bdring, 2021-04-11:
Depending on your probe type (normally open vs. closed) and circuitry, like specialized input module circuitry your probe might report inverted (active when it is not). The $6 allows you to compensate for that. That is way Arduino Grbl and Grbl_ESP32 have the $6 command. There is non standard value. You must set the value that works for your machine.

See this wiki page
https://github.com/bdring/Grbl_Esp32/wiki/Probe-Feature#probe-testing-and-setup

Teselation, 2021-04-11:
Well, my machine/probe didn't change but the $6 did it seems to me like Arduino's normally open state is $6 = 0 but ESP32's normally open state is $6 = 1 which (if im correct) is a mild inconvenience for someone moving from Arduino.

bdring, 2021-04-11:
What hardware are you using? Are you using a shield with the Arduino and you using something like a 6 pack and an input module for the ESP32?

Teselation, 2021-04-11:
### Arduino
[image]

### ESP32
[image]

In both cases probe pin is normally high.

For some reason i cant submit a new issue on Candle's forum.
[image]

Teselation, 2021-04-11:
Never mind about Candle's forum, I figured it out.

bdring, 2021-04-19:
Closing... This appears to be a Candle problem.

closed 2021-04-19 by bdring

papillion47, 2024-10-04:
Uzun zamandır Normal çalışmaktayken, GRBL ekranında visualizer işi ve işlemi göstermemeye başladı. Ayrıca Heightmap ayarlama kısmı da aktif değil, menüsü çalışmıyor. Sebep ne olabilir, çözümü nedir? Bir yardımda bulunulursa çok mutlu olacağım. Peşinen teşekkür ve selâmlarla.. (CNC ve bilgisayar yazılım ve donanım konularında acemiyim. Lütfen yardım ediniz.)

papillion47, 2024-10-04:
[image]