bdring/Grbl_Esp32 #863: Issues installing firmware with custom code to fresh ESP32

BarbourSmith, 2021-04-14:
First and foremost, I've been watching the project for over a year now and I'm a huge fan. The number of issues that people post which you respond to and solve is amazing. This project rocks.

Second, this is a bug in MY custom machine code, not the GRBL ESP32 project which has been rock solid for me. I'm just totally stumped at how I managed to break it in such a weird way and I'm hoping steer me in the right direction for where I went wrong.

This is a follow up to issue #330 

**What version of the firmware are you using?**

Grbl_ESP32 Ver 1.3a Date 20210401

**Is the problem repeatable?**

Yes

**Under what conditions does the bug occur?**

I am working on a closed loop controller built on GRBL ESP32. Basically there are DC motors with encoders on each axis instead of stepper motors. This requires a custom code file and a slight modification to `protocol_execute_realtime()`. 

It works great, GRBL ESP32 rocks...except that my firmware cannot be flashed to a fresh ESP32. If I upload the master branch firmware first and then my firmware afterwards all is well. If I try to upload my firmware to a fresh ESP32 it continuously core panics with error message.
```

[MSG:Grbl_ESP32 Ver 1.3a Date 20210401]
[MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
[MSG:Using machine:MASLOW 4]
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400df6cd  PS      : 0x00060930  A0      : 0x800d8a3c  A1      : 0x3ffd37d0
A2      : 0x3ffd3818  A3      : 0x3ffd380c  A4      : 0x00000000  A5      : 0x3ffd3824
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x00000000
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000001
A14     : 0x00060920  A15     : 0x00000000  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000

Backtrace: 0x400df6cd:0x3ffd37d0 0x400d8a39:0x3ffd37f0 0x400d8fd9:0x3ffd3880 0x400dbc07:0x3ffd38a0 0x400dbe1b:0x3ffd38c0 0x400dbe32:0x3ffd3900 0x400d7d5b:0x3ffd3950 0x400d4e77:0x3ffd3970 0x400d2dcf:0x3ffd3990 0x400fbcd3:0x3ffd39b0 0x400903c5:0x3ffd39d0
  #0  0x400df6cd:0x3ffd37d0 in FakeSetting<int>::get() at Grbl_Esp32/src/System.cpp:149
      (inlined by) system_convert_array_steps_to_mpos(float*, int*) at Grbl_Esp32/src/System.cpp:181
  #1  0x400d8a39:0x3ffd37f0 in protocol_execute_realtime() at Grbl_Esp32/src/Protocol.cpp:247
  #2  0x400d8fd9:0x3ffd3880 in protocol_buffer_synchronize() at Grbl_Esp32/src/Protocol.cpp:211
  #3  0x400dbc07:0x3ffd38a0 in Coordinates::set(float*) at Grbl_Esp32/src/Settings.cpp:752
  #4  0x400dbe1b:0x3ffd38c0 in Coordinates::setDefault() at Grbl_Esp32/src/Settings.h:244
      (inlined by) make_coordinate(CoordIndex, char const*) at Grbl_Esp32/src/SettingsDefinitions.cpp:238
  #5  0x400dbe32:0x3ffd3900 in make_settings() at Grbl_Esp32/src/SettingsDefinitions.cpp:246
  #6  0x400d7d5b:0x3ffd3950 in settings_init() at Grbl_Esp32/src/ProcessSettings.

```

The issue is my change to 

```C++

void protocol_execute_realtime() {
    
    int32_t current_position[N_AXIS]; // Copy current state of the system position variable
    memcpy(current_position, sys_position, sizeof(sys_position));
    float print_position[N_AXIS];
    system_convert_array_steps_to_mpos(print_position, current_position);
    
    recomputePID();
    setTargets(print_position[0], print_position[1], print_position[2]);
    
    protocol_exec_rt_system();
    if (sys.suspend.value) {
        protocol_exec_rt_suspend();
    }
}

```

Basically I'm recomputing the PID controllers for the axis positions and updating their target positions. Making this change to `Protocol.cpp` is what prevents me from being able to upload to a fresh ESP32.

Does anyone know why I've broken things in such a weird way or have any advice for how to fix them? I see the issue when compiling with either Platform.IO or the Arduino IDE.

Thanks!

MitchBradley, 2021-04-14:
Try undefining FORCE_BUFFER_SYNC_DURING_NVS_WRITE in Config.h

BarbourSmith, 2021-04-14:
Genius!

That fixed it completely. 

Any pointers on how you knew that was the right thing to do or what was going on?

closed 2021-04-14 by BarbourSmith

MitchBradley, 2021-04-15:
[image]
[image]

BarbourSmith, 2021-04-15:
Great catch :)

I wouldn't have seen that

MitchBradley, 2021-04-15:
You would if you had spent many hundreds of hours working on this code base.