bdring/Grbl_Esp32 #866: [MSG:Warning: Linear fit not implemented yet.]

Romain3481, 2021-04-16:
Hi there !

I'm trying Linear piecewise spindle, i set and ran fit_nonlinear_spindle.py, copy paste the solution in config.h, activated ENABLE_PIECEWISE_LINEAR_SPINDLE, but when i launch M3 S***** , i get [MSG:Warning: Linear fit not implemented yet.]
Thank for your help :)

`// Enables a piecewise linear model of the spindle PWM/speed output. Requires a solution by the
// 'fit_nonlinear_spindle.py' script in the /doc/script folder of the repo. See file comments
// on how to gather spindle data and run the script to generate a solution.
#define ENABLE_PIECEWISE_LINEAR_SPINDLE  // Default disabled. Uncomment to enable.

// N_PIECES, RPM_MAX, RPM_MIN, RPM_POINTxx, and RPM_LINE_XX constants are all set and given by
// the 'fit_nonlinear_spindle.py' script solution. Used only when ENABLE_PIECEWISE_LINEAR_SPINDLE
// is enabled. Make sure the constant values are exactly the same as the script solution.
// NOTE: When N_PIECES < 4, unused RPM_LINE and RPM_POINT defines are not required and omitted.

#define N_PIECES 4
#define RPM_MAX 24162.9
#define RPM_MIN 118.4
#define RPM_POINT12 3177.2
#define RPM_POINT23 13581.6
#define RPM_POINT34 21467.2
#define RPM_LINE_A1 5.878084e-01
#define RPM_LINE_B1 6.759473e+01
#define RPM_LINE_A2 9.803514e-01
#define RPM_LINE_B2 1.314789e+03
#define RPM_LINE_A3 1.014510e+00
#define RPM_LINE_B3 1.778715e+03
#define RPM_LINE_A4 1.483485e+00
#define RPM_LINE_B4 1.184633e+04
`

bdring, 2021-04-19:
That feature is not supported at this time. We have been talking about it on the Discord server.

closed 2021-04-19 by bdring

Romain3481, 2021-04-19:
Thank you for your reply