bdring/Grbl_Esp32 #869: Doubt in relation to uploading the Gcode files

AllanCosta12, 2021-04-18:
I was wondering if you can only upload the Gcode files via the SD module or have a way to download files that may be on my phone or computer;  thus connecting the device that has the file on the ESP network and uploading it

AllanCosta12, 2021-04-22:
I'm sorry if it's a lay question, but I didn't understand how I do it; I'm having more difficulties to move on the webUi, on the issue
of electronics and programming is fine, but then I wanted to know how can I do that, because when I enter WebUI, I only find to upload files "upload SD files", I would like to know more in detail how I upload via the web.
I saw a video I don't remember if it was yours, or some channel with a similar name, and trying to understand what it said it seems like I said I have to upload the file to the ESP flash, is that right? if not, would you be able to explain in detail how to upload the file without using SD?
thanks in advance.

MitchBradley, 2021-04-22:
It depends what you mean by "upload". One meaning is "transfer the file to either the SD card filesystem or the on-module ESP32 FLASH filesystem". Another meaning is "run the GCode file to move the machine"

When using the WebUI, you cannot run a GCode file unless it has first been stored on the SD filesystem. There is no way, using WebUI, to directly run a file from another computer without giong through the SD.

If you need to bypass the SD, you will have to use a conventional GCode sender program like UGS (there are many others besides UGS) that connects to Grbl_Esp32 via USB serial.

AllanCosta12, 2021-04-22:
in short when using webUi i need to have an SD card to make the machine run gcodes, if i want to run a gcode present on another device like a computer i need to connect with the serial port, correct?
is that I was unsure why you mentioned uploading it to the ESP32 flash.

closed 2021-04-22 by MitchBradley

luc-github, 2021-04-22:
@AllanCosta12 if you want to use gcode from another computer by WiFi you can use the TCP protocol or use Bluetooth in both cases you do not need the webui