bdring/Grbl_Esp32 #870: 3axis . After the control switch function is enabled in V4, all IO will not output

yeyoubin, 2021-04-19:
It uses the configuration of 3axis.v4 and compiles and uploads after the 34 35 36 39 pin function is enabled. After all the output IO did not respond, what is the reason? After commenting out these functions, other output IO returned to normal after compiling and uploading.Please instruct

Romain3481, 2021-04-19:
Hi,
Possible issue 1: The 34 35 36 39 pins don't have pull-up resistors, if you didn't put those resistors, you can't use those pins.
Wiki-> https://github.com/bdring/Grbl_Esp32/wiki/Setting-Up-the-I-O-Pins

Possible issue 2: If you put pull-up resistors as explained, it's possible that you have to invert control pin mask.

bdring, 2021-04-19:
Your description is very brief.  It is hard to know what you did or what you want to happen.

- What input functions do you want?
- What hardware are you using?
-  What type of switches are you using?
- Have you read this wiki page?
- What is the response the the status request command character '?'

yeyoubin, 2021-04-20:
It has been solved that I used the development board to test and did not connect the CNC expansion board, so 34 35 36 39 IOS did not pull up, resulting in other IOS did not output. Thank you for your guidance. I come from China. I don't know how to translate English words with translation software. I'm sorry for the lack of expression.

closed 2021-04-20 by bdring