bdring/Grbl_Esp32 #873: ESPDuino32 and CNC Shield

MilkLogger, 2021-04-20:
Hey guys
I bought a EspDuíno (Esp32) and I want to insert it into the CNC Shield just like you do with the arduino board. I did all the GRBL-ESP32 installation procedure, it is running fine, but every time I insert the CNC Shield, ESPDuíno cuts off the wifi signal that it emits and also cuts off the USB Serial communication with the program I use to machine the pieces called bCNC. If ESPDuíno is without the Shield attached, it works, that is, it emits the wifi signal and also the bCNC shows that it is connected.

Can you help me please?

[image]
[image]

joedirium, 2021-04-20:
Not sure if this is the solution, but I remember someone mentioned that the pull-up resistor on the shield must be removed to work with the ESP32 board.

BlueOrangeLive, 2021-04-20:
Look here.
https://github.com/bdring/Grbl_Esp32/issues/369#issue-593745142

bdring, 2021-04-20:
@MilkLogger 

If you get it working, can you please post a simple explanation? I will create some documentation to help the next person.

MilkLogger, 2021-04-21:
What de difference between ESP32-GRBL Master and ESP32-GRBL Main?

MitchBradley, 2021-04-21:
We renamed master to main a long time ago to be consistent with GitHub default usage - https://www.theserverside.com/feature/Why-GitHub-renamed-its-master-branch-to-main

Our daily development work goes into the Devt branch.  Every so often we promote Devt to main.  All pull requests should target Devt.  If you have a problem with main, try Devt.

MilkLogger, 2021-04-21:
I didn't find anywhere in the GRBL ESP32 files, the settings that are in the attachments.
Where can I find them to make the necessary changes?
[image]
[image]

bdring, 2021-04-23:
Homing cycles are adjustable as settings with $Homing/Cycle0=<axes>

Examples: 
- $Homing/Cycle0=Z
- $Homing/Cycle1=XY

You can define up to 6. Cycle0 through Cycle5

Set it to 0 or blank to remove axes from a cycle.

You can also set defaults in your machine definition like this

#define DEFAULT_HOMING_CYCLE_0   bit(Z_AXIS)

Is the original reason for this issue resolved? Please explain how you fixed you problem.
Do not use this issue for other questions.

runryder912, 2021-05-10:
As I was configuring the esp32_grbl file, I noticed that the reset pin resets the board and does not abort my cnc machine movement. How would I able to configure the reset pin to function as abort and trigger the alarm state to stop machine movement.

Thanks for  your help

dineshvyas, 2021-11-29:
There are many CNC shield versions are available if you try hookup ESP32 with CNC shield it wont work until you make connections directly to motor drivers header. there is something wrong with input pin sockets and motor drivers sockets.