bdring/Grbl_Esp32 #874: Setting up custom machine

xbavnu, 2021-04-20:
Hi,

I am currently working on a sand table (similar to the sisyphus table) but I am a bit lost when it comes to setting up the machine in Grbl. While I've done some Arduinos projects and worked with stepper motors, I've never worked with Grbl and dont't really know where to start and which things I have to configure in what way and what is necessary before flashing Grbl to the ESP32 and what can be changed afterwards etc.

I want to use the website www.sandify.org to create the GCode and then upload it to an SD-Card connected to the ESP32. So I think Grbl should be the ideal solution to do so.

The sand table uses one rotary axis (A) and one linear axis (X). The linear axis sits on top of the rotary axis (see the attached image). So all I really need, is to set up two axes and a limit switch for the linear axis. No spindle, no Z-axis etc. But at some point I have to tell Grbl that one axis is a rotary axis, input the gear reduction, probably do some conversions for the angles? Can these things be changed later on or do I need to flash Grbl again if I want to make changes to these things?

My idea was to use an existing machine profile and just make some changes to it. So I thought about using the "polar coaster" as a reference machine. However on the polar coaster the linear axis is fixed and not attached to the rotary axis. Does that even matter or shouldn't the result still be the same? Also the polar coaster uses a custom .cpp file, so it can not be uploaded via the Arduino IDE, right? 

Can somebody point me into the right direction on how and where to start?

[image]

[image]

bdring, 2021-04-20:
That looks really cool. 

It would be similar to the polar coaster. We are currently trying to simplify the kinematics for Grbl_ESP32. That should be done in a few days.

We have a Discord server with an **art-bots** channel. Here is an invite link.

https://discord.gg/UqMUQzdX3D 

You should join and post the question there.

bdring, 2021-04-20:
This might be a good controller for it. It really quiet with the TMC2209 drivers.

https://github.com/bdring/TMC2209_Pen_Laser

https://www.tindie.com/products/33366583/tmc2209-penlaser-controller/

xbavnu, 2021-04-21:
@bdring Thanks, I just posted the question there. Looking forward to some simplifications to the kinematics :D 

For the controller I want to repurpose a PCB I made recently, I just need to make some minor modifications to it for the SD-Card. For testing I will be using some A4988 drivers, but I already ordered some TMC2208 drivers.

jregholdsworth, 2021-05-23:
Trying to make a corexy laser engraver using GRBL esp32 and a Fystec 1.4 board. How do you alter the stepper driver current for the TMC 2209 chips?
And any hints on the corexy configuration?. I mashed up the Fystec machine file with the Midtbot machine file..seems to work..but is this the best approach?