bdring/Grbl_Esp32 #878: i2s does not work with external drivers

lnfactionfreddy, 2021-04-21:
I took 6_pack_TMC2209_XYZP_SpinRelay.h as the machine config.  Unfortunately, the steppers do not move at all and do not make any noise.  As a shift register I only use 74hc595n decency 74ahc595.  The driver is connected to Vm with 24v

lnfactionfreddy, 2021-04-22:
*6_pack_stepstick_v1.h the other was of the 6 Pack controller machine defs

jschoch, 2021-04-22:
it should work fine assuming your shift register is connected properly, can you put the input signal and output signals for the 74hc595n on a scope?

bdring, 2021-04-27:
Is this home built hardware or the 6 Pack Controller?

lnfactionfreddy, 2021-04-27:
It´s a home build Hardware. I have now solved it without shift register. Defining the board pins as dir and step worked. Thanks for the help.  I will soon find out Scope about the error.

bdring, 2021-04-27:
I'll close the issue. If you find anything helpful for us add it to the issue or reopen if necessary.

closed 2021-04-27 by bdring