bdring/Grbl_Esp32 #879: Add spindle function

yeyoubin, 2021-04-23:
[https://github.com/KnightSin/Grbl_Esp32](url)
Can the spindle function in this branch band code be integrated in the next update? When it is used in writing machine, electromagnet and rudder are added to the spindle. And when changing the spindle function, you only need to enter $spindle / type = en or $spindle / type = sever in the command bar

bdring, 2021-04-23:
Can you provide links to that spindle? I have never heard of it.

yeyoubin, 2021-04-25:
[https://detail.tmall.com/item.htm?id=530629424244&ali_refid=a3_430676_1006:1104520036:N:x2WZ9EGCA0Bbe/NuzYlpVQ==:70e8ac2953e6543d7cf5751e59eb4351&ali_trackid=1_70e8ac2953e6543d7cf5751e59eb4351&spm=a2e1u.19484427.29996460.24](url)  steering engine
[image]

[https://item.taobao.com/item.htm?id=601827000360&ali_trackid=2:mm_26632323_6762370_25910879:1619338376_145_461879816&union_lens=lensId:OPT@1619338360@212b7270_07e2_17908167897_ad10@01;recoveryid:201_11.23.72.131_522460_1619338254446;prepvid:201_11.1.81.73_631471_1619338360188&clk1=e329782072aa96ff6b4949e0d16b5101&spm=a2e1u.19484427.29996460.6&pvid=100_11.229.250.111_85960_1991619338360787765&scm=null&bxsign=tbkb5KuhqkaZBme963SSxpvv44ikiJIoOdXHc9q75Pdt3k5kB7wkUG/8B2Y6Q4UCCNZZPjF1m7T5vEB/FjYaF7KSeumFcnktYAiGR7DRP3Ma7w=](url)      electromagnt
[image]
Used fro pen lifting of writing machine

bdring, 2021-04-25:
We would not consider those spindles. We do have support for hobby servos as an axis motor. They are commonly used in pen plotters.

We could add an solenoid axis motor.

lkarlslund, 2021-06-04:
Solenoid with variable pressure would be great for vinyl cutters too - I think that's what many of them are using. But really it's just an analogue out (PWM) translated from GCode.