bdring/Grbl_Esp32 #880: Problems Compiling Firmware

Deltaplc, 2021-04-23:
Please answer the following questions:

Have you read the [wiki regarding how to compile](https://github.com/bdring/Grbl_Esp32/wiki/Compiling-the-firmware)?

What version of the Arduino IDE are you using? 

What version (commit date) of the [Arduino core for the ESP32](https://github.com/espressif/arduino-esp32) are you using? 

Are you using the master branch of Grbl_ESP32?

Have you made any edits or configuration changes (list them) to the  firmware?

Please paste the compiler error text here:

Deltaplc, 2021-04-23:
Have you read the wiki regarding how to compile? -- YES!!

- Arduino IDE 1.8.13
- ESP32 DEVKIT V1
_Arduino:1.8.13 (Windows 10), Scheda:"DOIT ESP32 DEVKIT V1, 80MHz, 921600, None"

ERRORE:
In file included from sketch\src/Grbl.h:54:0,

                 from C:\Users\Davide\Documents\Arduino\libraries\Grbl_Esp32\Grbl_Esp32.ino:21:

sketch\src/Uart.h:39:19: error: 'size_t Uart::readBytes(char*, size_t)' marked 'override', but does not override

     size_t        readBytes(char* buffer, size_t length) override;

                   ^

bdring, 2021-04-24:
It compiles fine for me. Are you sure you followed the Wiki exactly? What version of the ESP32 package are you using?

[image]

Deltaplc, 2021-04-24:
Thank you for the answer!
I tried everything again from the beginning on another pc and it worked!
Thank you
Greetings

closed 2021-04-24 by Deltaplc