bdring/Grbl_Esp32 #881: Uploading WebUI index.html.gz 'File write failed'

dleask, 2021-04-24:
I've complied and uploaded the latest build using both Platform.io and Arduino IDE and can get to the Web UI where it prompts "File index.html.gz is missing please upload it".
The Wiki points out that the file to be uploaded is located in Grbl_Esp32/src/data/index.html.gz but in the latest master the file 'data/index.html.gz' does not exist but there is a WebUi folder unfortunately the file is not in there either.
I've downloaded the latest file from https://github.com/luc-github/ESP3D-WEBUI/blob/2.1/index.html.gz but when I hit the upload button after selecting the file, the Web UI pops up an error box with "Update failed(3): File write failed". The downloaded file is 157Kb in size and the Web UI is showing Total space: 169.38KB/0B used.
The serial monitor is also showing '[MSG:Upload error]' when it fails.
Am I missing the correct location of the file to upload?

dleask, 2021-04-24:
My goof, the file was in Grbl_Esp32/data not Grbl_Esp32/src/data

The latest version from Luc's repo can't be substituted.

closed 2021-04-24 by dleask

bdring, 2021-04-24:
The folder moved recently

It is now here. I updated the wiki

https://github.com/bdring/Grbl_Esp32/tree/main/Grbl_Esp32/data

You should be able to upload it with this command "pio run -t uploadfs" from a platformio terminal

dleask, 2021-04-24:
Thanks for the swift response, all up and going now!