bdring/Grbl_Esp32 #884: Problems Compiling Firmware

bdanyeur, 2021-04-28:
Please answer the following questions:

Have you read the [wiki regarding how to compile](https://github.com/bdring/Grbl_Esp32/wiki/Compiling-the-firmware)? Yes several times.

What version of the Arduino IDE are you using?  1..8.13 window store 1.8.42.0

What version (commit date) of the [Arduino core for the ESP32](https://github.com/espressif/arduino-esp32) are you using? 
retrieved it from github earlier today
Are you using the master branch of Grbl_ESP32? Yes, at least I think so.  I'm an experienced programmer but a NOOB on arduino IDE

Have you made any edits or configuration changes (list them) to the  firmware?
no changes here
Please paste the compiler error text here:
C:\\Users\\Bill\\Documents\\Arduino\\hardware\\espressif\\esp32\\libraries\\WiFi\\src" "

C:\\Users\\Bill\\AppData\\Local\\Temp\\arduino_build_73837\\sketch\\src\\I2SOut.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE

Alternatives for FreeRTOS.h: []

C:\Users\Bill\AppData\Local\Temp\arduino_build_73837\sketch\src\I2SOut.cpp:54:10: fatal error: FreeRTOS.h: No such file or directoryResolveLibrary(FreeRTOS.h)

  -> candidates: [] #include <FreeRTOS.h>
          ^~~~~~~~~~~~
compilation terminated.

MitchBradley, 2021-04-28:
Instead of getting the Arduino core for ESP32from github, try [installing it using boards manager](https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/boards_manager.md) . We have recently experienced problems with bleeding-edge versions.  1.0.4 worked, then 1.0.5 broke the serial driver and the breakage persisted in 1.0.6.  In the latest Devt and main of grbl_esp32, we worked around that problem so 1.0.4, 1.0.5, and 1.0.6 all work, but we do not know what is in the not-yet-released code.

If you are comfortable with platformIO, you could try [compiling with PlatformIO](https://github.com/bdring/Grbl_Esp32/wiki/Compiling-with-PlatformIO) .  The developers all use PlatformIO so that gets tested every day, whereas Arduino IDE compiles are tested less frequently.

On my system, with Arduino IDE v 1.8.13 and Arduino Core for ESP32 version 1.0.6, the FreeRTOS.h file is present at /Users/wmb/AppData/Local/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/include/freertos/freertos/FreeRTOS.h

bdanyeur, 2021-04-29:
Thanks for your insight on the problem. I’ll try with Platform IO

  My problem started 5 days ago when I tried to use the board manager using the ESP download link  in preferences folder. I kept getting a download error using that method.  Tried several times. Always an error.  I’m still a Noob using Ubuntu so in case It was my error being unfamiliar with Ubuntu I switched over to using my familiar Windows 10 environment. I Got Same error in Windows.  
I followed some instructions in a Windows read me file using Git Bash and Git GUI. Now I have the boards listed in board manager.
I downloaded the GrBLESP32 zip and followed the instructions moving to the ARDUINO sketch.  
 I tried compiling using  the Grblesp32.INO and got the error.  All the other modules with references to the tools\SDK\include freertos look like they work.  
So yesterday I deleted everything including the ARDUINO IDE and started over again. New ARDUINO ide, preferences still would let me include the ESP links, so Git GUI,Git bash to get ARDUINO esp32 stuff. Downloaded a Fresh GRBLESP33.zip, followed the instructions.  

To insure I could even download to the board I used the esp sample Blink and successfully downloaded it to my Dev board. 
Guess it’s time to use platformio. I am going to upgrade my skills to include Python and C anyway. VSE and platformio means More learning. 
Thanks again. 
Bill
Sent from Bill’s iPhone

manotroll, 2021-06-05:
using the latest version that can be downloaded from the original page as zip does not appear on the arduino the .h files for configuration
https://i.imgur.com/lM5iOpg.png
but I can edit the .h files directly right?
and how to disable bluetooth and wifi ?
to save resources and free up more memory since I'm going to use cable

bdring, 2021-06-05:
Here are the compiling instructions.

https://github.com/bdring/Grbl_Esp32/wiki/Compiling-with-Arduino-IDE

You can disable Wifi and Bluetooth by editing the config.h file.

closed 2021-06-05 by bdring

manotroll, 2021-06-05:
I followed the tutorial and it works more like the picture above it doesn't show the .h files like in the tutorial image just the Grbl_ESP32

bdring, 2021-06-05:
You can edit the files in any text editor.

The versions of Arduino are probably different. Click the drop down button in the upper right.