bdring/Grbl_Esp32 #885: JMC IHSV57 step per mm

qiwalker, 2021-04-29:
Hi

I use JMC step servo with 800 pulse per rev
How to setting step per mm and also suitable pulse

Thanks

bdring, 2021-04-29:
There is information on the step and direction pulse options here.

https://github.com/bdring/Grbl_Esp32/wiki/External-Stepper-Drivers

There is a basic setting for steps/mm $<axis>/StepsPerMm=

https://github.com/bdring/Grbl_Esp32/wiki/Settings#xstepspermm-thru-cstepspermm-or-100-thru-105--xyzabc-stepsmm

The wiki has all the information you asked for.

https://github.com/bdring/Grbl_Esp32/wiki

qiwalker, 2021-04-29:
Thankyou so much
But in this motor i cant found microstepping only table of pulse per rev

bdring, 2021-04-29:
That generally assumes 200 full steps per rev, so 800 would be 4x micro stepping.

qiwalker, 2021-04-29:
Thanks for ur explanation
 Change the step per mm value make DRO show wrong value, 
Ie : jog machine 1mm then the DRO display is 0.999

Anyway how to set machine to zero after homing ?

bdring, 2021-04-29:
The machine moves in steps. It might not end up exactly where you want, but 0.001mm is **really, really** close. 

It is better for the machine to say exactly where it is.

You can zero anywhere you want.

https://www.youtube.com/watch?v=fGtbkVJBXyE

Most senders have button to zero the axes as well.

closed 2021-05-04 by bdring