bdring/Grbl_Esp32 #887: Use Servo on only one axis

AllanCosta12, 2021-05-04:
I was wondering if I can use a servo instead of a motor on the Z axis, the wiki is saying that I need to go to the machine's configuration file, I was unsure of what that file would be, so I looked at some, like: "config .h", "Machine.h" and I didn't find anything, so I went to the "motors" folder and looking at some files I saw that it has a hierarchical order, it has the "MOTOR_USED" file, which belongs to "Motor" which in turn belongs to "Motors"; But seeing this I had two doubts:

1- If I use 'RcServo.h' which would be the library to use the servo, would I be obliged to use the servo in the 3 axes? How can I use the servo on just one axis?
2- How would you do to 'link' all this order in the schedule, from which header file would I do this?

NOTE: I will use these simple servos, follow the name below:
servo motor 9g sg90

I hope you understand.

bdring, 2021-05-04:
Here is an example machine definition file that uses a servo Z

https://github.com/bdring/Grbl_Esp32/blob/main/Grbl_Esp32/src/Machines/midtbot.h

AllanCosta12, 2021-05-05:
I took it, ended up taking a look and I hadn't found the file.
But now I'm having another problem, when defining the "midtbot" machine in the "machine" file, the servo and motors freeze, I try to move it by hand, but it is engaged, I tried to change the code but I don't know what can to be.
as the engines freeze and the drivers heat up, the GPIOS connection is correct, as there is a signal going to the drivers and motors, so I think it could be something in the programming.
NOTE: I'm on the prototype, I don't have a limit switch, it's just on the protoboard.
Sorry for the lay question.

bdring, 2021-05-05:
The servos should lock just like steppers when the motor enable is on. Read this section in the wiki for more information 

https://github.com/bdring/Grbl_Esp32/wiki/Servo-Axis-Feature

The Wiki can answer some of your other questions as well.

AllanCosta12, 2021-05-06:
thank you, everything resolved.

closed 2021-05-06 by AllanCosta12