bdring/Grbl_Esp32 #890: Multi stepper Y-axis squaring  issue

bhboyle, 2021-05-07:
Good afternoon folks.

I have a question. I have built a new CNC using ESP32 grbl and i may have done something bad. I have two steppers for the Y-axis and two limit switches, one for each side of the Y-axis. the two steppers are driven on thier own GPIO pin and so are the two limit switches. Each has its own GPIO pin. how ever I am seeing an issue with the Y-2 squaring during homeing. When I start a homing X ans Z are both working fine. Then the Y starts and Y-1 works perfect. But when the system trys to square and Home the Y-2 channel it does not work and I have to power the system down due to the stepper being over driven.

Is this because I have two limit pins? They are defined in my config and it compiled and installed correctly.  I have also cnofimed that the limit switch is correctly triggering the second limit pin with a multimeter.

Do I need to have both limits go to one pin?

Thank you in advance for any help you may provide.

bdring, 2021-05-08:
You can use 1 or 2 switches per axis. Squaring will work with either setup. 

You should verify that all switches work by testing them manually before trying to home.

https://github.com/bdring/Grbl_Esp32/wiki/Help-with-Switch-Problems

bhboyle, 2021-05-08:
Thank you very much for your help.

I have reconfirmed the triigers work as expected with my multi-meter right at the GPIO pins (check both Y-1 and Y-2 seperately) on the ESP32 and via the web interface console output that showed

<Run|MPos:351.154,0.000,-135.983,10.477|FS:1200,0|Pn:Y>

for both the Y-1 and Y-2 triggers. However, the squaring still does not work correctly. The Y-2 squaring causes the stepper to over drive and I have to kill the power.

BUT, this morning I added an And Gate for the limit switches (they are optical) and brought both switches into the Y-1 GPIO input, the problem went away and squaring works as expected. This is fine for me it just means I need a little daughter board for the And Gate.

Cheers.

MitchBradley, 2021-05-08:
You can wire-AND the switches if they have the usual NPN open-collector outputs, eliminating the need for the daughter card.

bhboyle, 2021-05-08:
Before anyone looks at any code.... This soultion did not resolve my problem. Still investigating but still having the same problem even with the AND Gate.

bhboyle, 2021-05-09:
So please accept my most humble of apoligies. The probelm was caused by the sun interfering with my optical sensors. I had the garage door open and the sensor closest to the door was the problematic sensor. I should have thought of this before. 

Thanks for all your help and the great work on the software!

closed 2021-05-09 by bdring