bdring/Grbl_Esp32 #892: reset/abort function gpio2

runryder912, 2021-05-10:
As I was configuring the esp32_grbl file, I noticed that the reset pin resets the board and does not abort my cnc machine movement. How would I able to configure the reset pin to function as abort and trigger the alarm state to stop machine movement.

Thanks for  your help

bdring, 2021-05-10:
Info about control switches is here.

https://github.com/bdring/Grbl_Esp32/wiki/Control-Switches

Make sure they are reporting their state in idle before you try testing with motion.

https://github.com/bdring/Grbl_Esp32/wiki/Help-with-Switch-Problems#checking-switch-status

If you still have problems, please follow this guide to give us information to help solve the problem.

runryder912, 2021-05-10:
I did some testing with an arduino uno with grbl 1.1. It is responding the
way I like. IT stops the program with a short of pin A0 with Alarm 3. This
is what I was looking for when I press the estop with the reset/about pin
shorted to ground. This kills program. Is this possible with an esp32
[image: image.png]
[image: image.png]

MitchBradley, 2021-05-11:
You need to provide the information listed in https://github.com/bdring/Grbl_Esp32/wiki/Requesting-Help .  Without that information, we cannot deduce what might be going wrong in your situation.

Protoneer, 2021-05-13:
I am having similar issue...

**I define my pin as follow:**
`
#define CONTROL_RESET_PIN       GPIO_NUM_27 
`

Then when I press the button my ESP32 restarts. (I start a movement first G0X1000)

**This is the message I get:**

Grbl 1.3a ['$' for help]
<Idle|MPos:0.000,0.000,0.000,0.000,0.000,0.000|FS:0,0|WCO:0.000,0.000,0.000,0.000,0.000,0.000>
<Idle|MPos:0.000,0.000,0.000,0.000,0.000,0.000|FS:0,0|Ov:100,100,100>
<Idle|MPos:0.000,0.000,0.000,0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000,0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000,0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000,0.000,0.000,0.000|FS:0,0>
ok
[MSG:Reset via control pin]
[MSG:Reset via control pin]
[MSG:Reset via control pin]
ALARM:3
[MSG:Reset via control pin]
[MSG:Reset Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Core 1 register dump:
PC      : 0x400951d2  PS      : 0x00060334  A0      : 0x800943e7  A1      : 0x3ffc0600
A2      : 0x3ffd583c  A3      : 0x3ffbced4  A4      : 0x00000001  A5      : 0x00000001
A6      : 0x00060323  A7      : 0x00000000  A8      : 0x3ffbced4  A9      : 0x3ffbced4
A10     : 0x00000018  A11     : 0x00000018  A12     : 0x00000001  A13     : 0x00000001
A14     : 0x00060321  A15     : 0x00000000  SAR     : 0x0000001a  EXCCAUSE: 0x00000006
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffff8
Core 1 was running in ISR context:
EPC1    : 0x400938ca  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400951d2

ELF file SHA256: 0000000000000000

Backtrace: 0x400951d2:0x3ffc0600 0x400943e4:0x3ffc0620 0x40092677:0x3ffc0640 0x401706e4:0x3ffc0680 0x40171091:0x3ffc06c0 0x400df9ad:0x3ffc06f0 0x400da652:0x3ffc0710 0x400d8cd6:0x3ffc0730 0x400d8d8a:0x3ffc07d0 0x400df7bc:0x3ffc0890 0x400821b0:0x3ffc08b0 0x400822ad:0x3ffc08d0 0x4008d899:0x3ffc08f0 0x4000bfed:0x3ffbcc50 0x40093a39:0x3ffbcc60 0x40092623:0x3ffbcc80 0x401ed5fe:0x3ffbccc0 0x401dcf69:0x3ffbcce0 0x400efe2e:0x3ffbcd10 0x400f73e1:0x3ffbcd50 0x400e2b49:0x3ffbcd80 0x400eb5a2:0x3ffbcdc0 0x400eb25c:0x3ffbcde0 0x400da5a7:0x3ffbce00 0x400928de:0x3ffbce20

Core 0 register dump:
PC      : 0x400938ca  PS      : 0x00060034  A0      : 0x800940ad  A1      : 0x3ffc0270
A2      : 0x3ffbf8c8  A3      : 0x0000cdcd  A4      : 0xb33fffff  A5      : 0x00000001
A6      : 0x00060021  A7      : 0x0000abab  A8      : 0x0000abab  A9      : 0x3ffc8fa0
A10     : 0x00000003  A11     : 0x00060023  A12     : 0x00060021  A13     : 0x0000001a
A14     : 0x0000002a  A15     : 0x3ffde750  SAR     : 0x00000018  EXCCAUSE: 0x00000006
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

ELF file SHA256: 0000000000000000

Backtrace: 0x400938ca:0x3ffc0270 0x400940aa:0x3ffc02a0 0x400929cb:0x3ffc02c0 0x40095051:0x3ffc02e0 0x4008d8a2:0x3ffc02f0 0x40222bd7:0x3ffbd540 0x4018bfb2:0x3ffbd560 0x40094075:0x3ffbd580 0x400928de:0x3ffbd5a0

Rebooting...

Protoneer, 2021-05-13:
I also tried the abort button while in idle mode with the same results.

bdring, 2021-05-13:
@Protoneer   I think there is some bounce on the reset pin which is causing the the interrupt to be called many times. The interrupt is printing the message `[MSG:Reset via control pin]`. 

Printing from an interrupt is risky and the multiple hits is probably causing the crash. @MitchBradley has identified a few other areas of concern in the control pin handling and started some code changes.

If you want to try some things on your own before his code is merged..
- Remove `grbl_msg_sendf(CLIENT_SERIAL, MsgLevel::Info, "Reset via control pin");` in System.cpp
- Try adding some RC filtering to the switch

Protoneer, 2021-05-13:
Thanks @bdring ,

I will test the filter and the code changes tonight.

Protoneer, 2021-05-18:
@bdring , I did some testing again and #901 has fixed the reboot issue. 

I have tested the code changes without adding a pull resister and when the button bounces a bit all I get is a repeat of the unlock message. (I am currently using my tweezer to short the button pins... :) )

```
<Run|MPos:165.830,0.000,0.000,0.000,0.000,0.000|FS:1000,0|Pn:XYZABC|WCO:0.000,0.000,0.000,0.000,0.000,0.000>
<Run|MPos:168.220,0.000,0.000,0.000,0.000,0.000|FS:1000,0|Pn:XYZABC|Ov:100,100,100>
ALARM:3

Grbl 1.3a ['$' for help]
[MSG:'$H'|'$X' to unlock]

Grbl 1.3a ['$' for help]
[MSG:'$H'|'$X' to unlock]

Grbl 1.3a ['$' for help]
[MSG:'$H'|'$X' to unlock]

Grbl 1.3a ['$' for help]
[MSG:'$H'|'$X' to unlock]

Grbl 1.3a ['$' for help]
[MSG:'$H'|'$X' to unlock]

```