bdring/Grbl_Esp32 #894: Laser output pin Issue

zerosha73, 2021-05-12:
This is my first try at customizing my own machine definition.  I keep getting [MSG:Warning: LASER_OUTPUT_PIN not defined] error no matter what I try.  It appears I have it defined below.  I can turn my laser on using M3 command, but any M4 or Spindle/Type=laser results in this error.  Running on a Wemos D1 R32 controller w/CNCShield 3.51.  I don't know if i am missing a pin definition somewhere.  

I would appreciate any assistance.  Here is my Machine definition text.

#define MACHINE_NAME            "ESP32_V4"

#define X_STEP_PIN              GPIO_NUM_26
#define X_DIRECTION_PIN         GPIO_NUM_16
#define Y_STEP_PIN              GPIO_NUM_25
#define Y_DIRECTION_PIN         GPIO_NUM_27
#define Z_STEP_PIN              GPIO_NUM_17
#define Z_DIRECTION_PIN         GPIO_NUM_14

#define X_LIMIT_PIN             GPIO_NUM_13
#define Y_LIMIT_PIN             GPIO_NUM_5
#define Z_LIMIT_PIN             GPIO_NUM_19

// OK to comment out to use pin for other features
#define STEPPERS_DISABLE_PIN    GPIO_NUM_12

#define SPINDLE_TYPE SpindleType::LASER
#define LASER_OUTPUT_PIN GPIO_NUM_23  // laserPWM
#define LASER_ENABLE_PIN GPIO_NUM_18  // LaserEnable

#define SPINDLE_TYPE            SpindleType::PWM
#define SPINDLE_OUTPUT_PIN      GPIO_NUM_23  // labeled SpinPWM
#define SPINDLE_ENABLE_PIN      GPIO_NUM_18  // labeled SpinEnbl

#define COOLANT_MIST_PIN        GPIO_NUM_34  // labeled Mist
// #define COOLANT_FLOOD_PIN       GPIO_NUM_39  // labeled Flood
#define PROBE_PIN               GPIO_NUM_39  // labeled Probe

sjonholle, 2021-05-12:
maybe this will help:
// #define SPINDLE_TYPE SpindleType::PWM
// #define SPINDLE_OUTPUT_PIN GPIO_NUM_23 // labeled SpinPWM
// #define SPINDLE_ENABLE_PIN GPIO_NUM_18 // labeled SpinEnbl

zerosha73, 2021-05-12:
I tried that already.  Still got the same error.  My understanding is you can have multiple spindle types defined and select the option you are using by selecting it with the "$Spindle/Type=xxxxx" command.  So the above should not be the issue. I am using a laser/router combo on my machine.  I tried to create a pure laser machine before coming here.

bdring, 2021-05-12:
You cannot define SPINDLE_TYPE twice. You actually don't need to define it at all if you plan to change it at run time.

Other than that, your machine definition works fine for me.

Are you sure you properly included it in Machine.h? Paste your boot messages in a reply.

https://github.com/bdring/Grbl_Esp32/wiki/Requesting-Help#boot-messages

zerosha73, 2021-05-12:
Ok, So i need to remover the "#define SPINDLE_TYPE SpindleType::PWM".  I understand my error there.

Here is my boot message;

[MSG:Grbl_ESP32 Ver 1.3a Date 20210424]
[MSG:Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:Using machine:ESP32_V4]
[MSG:Axis count 3]
[MSG:RMT Steps]
[MSG:Init Motors]
[MSG:Global stepper disable pin:GPIO(12)]
[MSG:X  Axis Standard Stepper Step:GPIO(26) Dir:GPIO(16) Disable:None Limits(0.000,730.000)]
[MSG:Y  Axis Standard Stepper Step:GPIO(25) Dir:GPIO(27) Disable:None Limits(0.000,625.000)]
[MSG:Z  Axis Standard Stepper Step:GPIO(17) Dir:GPIO(14) Disable:None Limits(-165.000,0.000)]
[MSG:Warning: LASER_OUTPUT_PIN not defined]

[MSG:Local access point GRBL_ESP started, 192.168.0.1]
[MSG:Captive Portal Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]
[MSG:Mist coolant on pin GPIO(34)]
[MSG:X  Axis limit switch on pin GPIO(13)]
[MSG:Y  Axis limit switch on pin GPIO(5)]
[MSG:Z  Axis limit switch on pin GPIO(19)]
[MSG:Probe on pin GPIO(39)]

zerosha73, 2021-05-12:
Here is my Machine.h file.  That is my custom file name.
#ifndef MACHINE_FILENAME

// !!! For initial testing, start with test_drive.h which disables
// all I/O pins
// #include "Machines/atari_1020.h"
// #    include "3axis_v4.h"

// !!! For actual use, change the line above to select a board
// from Machines/, for example:
#include "Machines/ChrisLaz.h"

bdring, 2021-05-12:
It works fine for me with your machine definition. You must have something setup wrong.

zerosha73, 2021-05-12:
Thank you. 

I will try and start with a directory refresh and recreate the file.  Will not rule out I messed something up somewhere.  At least I know my definition file is done correctly.

zerosha73, 2021-05-19:
Thank you @bdring.  What might be rudimentary to a seasoned programmer, I made the mistake of copying folders and my files were not in same location I had originally launched the .ino file from.  So it kept looking at the original files and not the ones I was modifying.